open62541/open62541 #2373: Access violation when running tutorial_server_firststeps and MinSizeRel build on Windows

Eebo78, 2019-01-17:
## Description
Building the examples using the MinSizeRel build configuration in Visual Studio 2017 the tutorial_server_firststeps.exe crashes. Running the sample from VS throws an access violation exception. Debug and Release configurations works fine. Same issue on win32 and x64.

## Background Information / Reproduction Steps
1. Download the sourcecode
2. Build for windows according to documentation. Enable Shared Libs, Build Exception and Amalgamation option.
3. Open VS and build ALL_BUILD with MinSizeRel configuration
4a. Copy the open62541.dll into the example folder
4b. Run tutorial_server_firststeps.exe. This crashes with a badtypemissmatch message in console
5a. Create a new C++ console and configure it to use the open62541 library
5b. Create a simple server.cpp and add code from tutorial
5c. Start debug session and the debugger throws a Exception Unhandled exception at 0x00... (open62541.dll) Access violation writing location. Locals window have following value on retval: Variable is optimized away and not available

## Checklist

Please provide the following information:

 - [x ] open62541 Version (release number or git tag):0.3.0
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x ] Operating system: Windows 10 64bit
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2019-01-17:
Interesting behaviour...
I guess VS2017 is optimizing some stuff which it should not do.

There are two ways to find out where the issue occurs:

1. Maybe you can generate somehow debug symbols together with the MinSizeRel build to be able to see where the exception happens. Maybe https://docs.microsoft.com/en-us/cpp/build/reference/how-to-debug-a-release-build?view=vs-2017

2. If you can not generate debug symbols, you can use printf or logging, step by step adding it to the code to find the last output before it crashes. Maybe setting the log level to TRACE also helps to find the line

teccheck, 2025-04-05:
As there has been no further reply for quite some time, I believe this is not relevant anymore. Also, a lot has changed since then, so I'll close this issue

closed 2025-04-05 by teccheck