open62541/open62541 #2377: BadCertificateHostNameInvalid

meitf, 2019-01-18:
## Description
I use the example download from the offical web ("https://open62541.org/"), I launch the windows server, and connet the server with  opc fundation client, it thrown exception of "BadCertificateHostNameInvalid", it also happend the same exception when i compile the server_ctt code of github and connet, but it is running well when i use the linux bin file server_ctt, client also connetc very well.
## Background Information / Reproduction Steps
1、compile the windows server_ctt and run or directly run the download example of 0.3 version bin server_ctt file
2、it comes "[2019-01-18 20:35:24.269 (UTC+0800)] warn/userland      Using server without certificate
[2019-01-18 20:35:24.426 (UTC+0800)] warn/network       Error binding a server socket: 以一种访问权限不允许的方式做了一个访问套接字的尝试。
[2019-01-18 20:35:24.430 (UTC+0800)] warn/network       Error binding a server socket: 以一种访问权限不允许的方式做了一个访问套接字的尝试。
[2019-01-18 20:35:24.435 (UTC+0800)] info/network       TCP network layer listening on opc.tcp://meitf-sz:4840/" 
3、connect the server with opc fundation client, it throw "BadCertificateHostNameInvalid" error, detail is "在 Opc.Ua.Client.Session.CheckCertificateDomain(ConfiguredEndpoint  endpoint)在 Opc.Ua.Client.Session.Create(ApplicationConfiguration  configuration, ConfiguredEndpoint endpoint, Boolean updateBeforeConnect, Boolean  checkDomain, String sessionName, UInt32 sessionTimeout, IUserIdentity identity,  IList`1 preferredLocales)在  Opc.Ua.Client.Controls.ConnectServerCtrl.Connect()在  Opc.Ua.Client.Controls.ConnectServerCtrl.Server_ConnectMI_Click(Object sender,  EventArgs e)
--
"
4、launcher by linux the functions will be ok 

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
0.3
 - [ ] Other OPC UA SDKs used (client or server): 
server
 - [ ] Operating system:
windows and linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Infinity95, 2019-01-18:
Hello,
I would guess that the server sockets you want to bind to are already bound by a different application.
Unfortunately i cannot read Chinese. Would it be possible for you to translate the error messages to english, or set your system to output the english error messages?
You could also check if another program is already using the port by running the following powershell command:
```Get-Process -Id (Get-NetTCPConnection -LocalPort portNumber).OwningProcess```
But remember to replace portNumber with the port you are trying to bind to.

marwinglaser, 2025-02-25:
Seems to no longer be relevant.

closed 2025-02-25 by marwinglaser