open62541/open62541 #2383: Implementation of FileType and friends

basyskom-meerkoetter, 2019-01-21:
I've done a (partial) implementation of the functionality required for FileType (open, read, close, size, etc...) on Linux. See Part5, Appendix. I've done this in the scope of a server application. I wonder if this isn't something that could be part of open62541? This isn't something that everybody that requires a FileType should implement over and over again.
I wonder where this would fit in? Is this a plugin?

Pro, 2019-01-21:
I guess it better fits into the arch folder since file operations are heavily dependent on the used architecture.
You can add separate files in the arch/posix folder and to the corresponding CMake

basyskom-meerkoetter, 2019-01-23:
Sounds good

chruetli, 2020-01-31:
@basyskom-meerkoetter: Can you pudh or somwhoe share your source? I'm looking for the filetransfer too!

EmbSys2019, 2020-05-11:
@basyskom-meerkoetter hello, can you plz provide us with an example how to do it?

tsillanp, 2020-07-13:
Hi @basyskom-meerkoetter any possibility to share the FileType -addon? Seems that this is now rewritten by multiple persons, but no commits yet to the repo: https://github.com/open62541/open62541/issues/3668

masongk, 2021-01-28:
@basyskom-meerkoetter Is it possible to share the branch with this feature?

basyskom-meerkoetter, 2021-01-28:
Unfortunately this code is not available. I am sorry.

basyskom-meerkoetter, 2021-01-28:
But judging from the comments on the ticket I see the interest in having this available in the open

samAvr, 2021-07-14:
@opcua-tsn-team-kalycito using your implementation works, if a small file will be uploaded from the pc to the uC .. the uC must have enough free RAM to be able to receive the file.
In the write callback function, `input[1].data;` will hold the wohle received data and if the RAM is not big enough, it causes in my case HardFault.
Is there any way to receive the file in small pieces and after that handle it?