open62541/open62541 #2387: Create event failed

FishEatingCat, 2019-01-24:
## Description
I used [https://github.com/open62541/open62541/blob/master/examples/tutorial_server_events.c](url) to test the event, but prompted me to report an error.

## Background Information / Reproduction 
Tools and platforms：
Windows 10 home
Microsoft Visual Studio 2017

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

```
Severity Code Description Item File Line Prohibit display status
Error LNK2019 Unresolved external symbol _UA_StatusCode_name, which is referenced in function _generateEventMethodCallback Project6 F:\work\CodeSource\OPCUA\open62541VSTest - copy\Project6\Project6\main.obj 1
Error LNK2001 Unresolved external symbol _UA_StatusCode_name Project6 F:\work\CodeSource\OPCUA\open62541VSTest - Copy\Project6\Project6\ua_client.obj 1
Error LNK2001 Unresolved external symbol _UA_StatusCode_name Project6 F:\work\CodeSource\OPCUA\open62541VSTest - Copy\Project6\Project6\ua_client_subscriptions.obj 1
Error LNK2001 Unresolved external symbol _UA_StatusCode_name Project6 F:\work\CodeSource\OPCUA\open62541VSTest - Copy\Project6\Project6\ua_server.obj 1
Error LNK2001 Unresolved external symbol _UA_StatusCode_name Project6 F:\work\CodeSource\OPCUA\open62541VSTest - Copy\Project6\Project6\ua_subscription_events.obj 1
Error LNK2019 Unresolved external symbol _UA_DateTime_now, this symbol is referenced in function _UA_Subscription_deleteMembers Project6 F:\work\CodeSource\OPCUA\open62541VSTest - copy\Project6\Project6\ua_subscription.obj 1
Error LNK2001 Unresolved external symbol _UA_DateTime_now Project6 F:\work\CodeSource\OPCUA\open62541VSTest - Copy\Project6\Project6\ua_subscription_events.obj 1
Error LNK2001 Unresolved external symbol _UA_DateTime_now Project6 F:\work\CodeSource\OPCUA\open62541VSTest - Copy\Project6\Project6\ua_types.obj 1
Error LNK2001 Unresolved external symbol _UA_DateTime_now Project6 F:\work\CodeSource\OPCUA\open62541VSTest - Copy\Project6\Project6\main.obj 1
Error LNK2001 Unresolved external symbol _UA_DateTime_now Project6 F:\work\CodeSource\OPCUA\open62541VSTest - Copy\Project6\Project6\ua_client.obj 1
```
Of course, this is part of the error message, but it is all this kind of error. I want to know what library is called by event? Otherwise why is this error?

I used the routines such as toturial_server_variable directly before.

FishEatingCat, 2019-01-25:
By compiling the open62541-master file with the previous single file collection, I have compiled the tutorial_server_evenr.c and built the server.
However, after successfully connecting using UA Export, calling Generate Event does not successfully create an event.

```
[2019-01-25 15:00:31.613 (UTC+0800)] info/session       Connection 0 | SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | AddNodes: Parent node i=3035 not found
[2019-01-25 15:00:31.614 (UTC+0800)] info/session       Connection 0 | SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | AddNodes: The parent reference for i=4035 is invalid with status code BadParentNodeIdInvalid
[2019-01-25 15:00:31.614 (UTC+0800)] info/session       Connection 0 | SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | AddNodes: Parent node i=2041 not found
[2019-01-25 15:00:31.614 (UTC+0800)] info/session       Connection 0 | SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | AddNodes: The parent reference for i=2713092980 is invalid with status code BadParentNodeIdInvalid
[2019-01-25 15:00:31.862 (UTC+0800)] info/network       TCP network layer listening on opc.tcp://DESKTOP-KVNLBPK:4840/
[2019-01-25 15:00:33.621 (UTC+0800)] info/network       Connection 632 | New connection over TCP from 192.168.31.139
[2019-01-25 15:00:33.622 (UTC+0800)] info/channel       Creating a new SecureChannel
[2019-01-25 15:00:33.622 (UTC+0800)] warn/securitypolicy        No PKI plugin set. Accepting all certificates
[2019-01-25 15:00:33.622 (UTC+0800)] info/channel       Connection 632 | SecureChannel 1 | Opened SecureChannel
[2019-01-25 15:00:33.623 (UTC+0800)] info/channel       Connection 632 | SecureChannel 1 | CloseSecureChannel
[2019-01-25 15:00:33.623 (UTC+0800)] info/network       Connection 632 | Closed
[2019-01-25 15:00:33.627 (UTC+0800)] info/network       Connection 636 | New connection over TCP from 192.168.31.139
[2019-01-25 15:00:33.627 (UTC+0800)] info/channel       Creating a new SecureChannel
[2019-01-25 15:00:33.627 (UTC+0800)] warn/securitypolicy        No PKI plugin set. Accepting all certificates
[2019-01-25 15:00:33.628 (UTC+0800)] info/channel       Connection 636 | SecureChannel 2 | Opened SecureChannel
[2019-01-25 15:00:33.628 (UTC+0800)] info/session       Connection 636 | SecureChannel 2 | Session ns=1;g=d79f8c57-a0db-d8d9-ee3e-c3dc3ed39d79 | ActivateSession: Session activated
[2019-01-25 15:00:33.634 (UTC+0800)] info/session       Connection 636 | SecureChannel 2 | Session ns=1;g=d79f8c57-a0db-d8d9-ee3e-c3dc3ed39d79 | Subscription 1 | Created the Subscription with a publishing interval of 100.00 ms
[2019-01-25 15:00:33.635 (UTC+0800)] info/session       Connection 636 | SecureChannel 2 | Session ns=1;g=d79f8c57-a0db-d8d9-ee3e-c3dc3ed39d79 | Subscription 1 | MonitoredItem 1 | Created the MonitoredItem
[2019-01-25 15:00:33.635 (UTC+0800)] info/session       Connection 636 | SecureChannel 2 | Session ns=1;g=d79f8c57-a0db-d8d9-ee3e-c3dc3ed39d79 | Subscription 2 | Created the Subscription with a publishing interval of 500.00 ms
[2019-01-25 15:00:33.636 (UTC+0800)] info/session       Connection 636 | SecureChannel 2 | Session ns=1;g=d79f8c57-a0db-d8d9-ee3e-c3dc3ed39d79 | Subscription 2 | MonitoredItem 1 | Created the MonitoredItem
[2019-01-25 15:00:33.636 (UTC+0800)] info/session       Connection 636 | SecureChannel 2 | Session ns=1;g=d79f8c57-a0db-d8d9-ee3e-c3dc3ed39d79 | Subscription 2 | MonitoredItem 2 | Created the MonitoredItem
[2019-01-25 15:00:33.637 (UTC+0800)] info/session       Connection 636 | SecureChannel 2 | Session ns=1;g=d79f8c57-a0db-d8d9-ee3e-c3dc3ed39d79 | Subscription 2 | MonitoredItem 3 | Created the MonitoredItem
[2019-01-25 15:00:33.638 (UTC+0800)] info/session       Connection 636 | SecureChannel 2 | Session ns=1;g=d79f8c57-a0db-d8d9-ee3e-c3dc3ed39d79 | Subscription 2 | MonitoredItem 4 | Created the MonitoredItem
[2019-01-25 15:00:38.290 (UTC+0800)] info/userland      Creating event
[2019-01-25 15:00:38.290 (UTC+0800)] error/userland     Event type must be a subtype of BaseEventType!
[2019-01-25 15:00:38.292 (UTC+0800)] warn/server        createEvent failed. StatusCode BadInvalidArgument
[2019-01-25 15:00:38.292 (UTC+0800)] warn/userland      Creating event failed. StatusCode BadInvalidArgument
```

basyskom-meerkoetter, 2019-02-07:
I can not reproduce your issue (master branch).

Seup:
./tutorial_server_events and ./tutorial_client_events are running an connected.
I connect via UaExpert and call the eventTrigger method.
The client will receive the event.

Observation: the event is not showing up in the UaExpert event view. Any ideas why?

GlorySky, 2019-02-21:
> 通过使用先前的单个文件集合编译open62541-master文件，我编译了tutorial_server_evenr.c并构建了服务器。
> 但是，使用UA Export成功连接后，调用Generate Event无法成功创建事件。
[2019-02-21 14:01:08.934 (UTC+0800)] info/userland	Creating event
1,090D1DE8,0 (EventNode info)

[2019-02-21 14:01:20.954 (UTC+0800)] error/userland	Event type must be a subtype of BaseEventType!

FishEatingCat, 2019-02-22:
hi.
I have not solved this problem yet.
Due to other projects, the OPC UA progress has temporarily stopped.

Good luck! @GlorySky

GlorySky, 2019-02-22:
Thanks for your reply,
Yesterday I had a lot of debugging and diagnostics and found that it might be because we didn't add a response to **BaseEventType** that caused the problem, and from some previous problem records, nodes such as **BaseEventType** had been defined when the server was built. and I cannot find the **BaseEventType** if only use the open62541.h/c or the main branch code.
looking at [](https://github.com/open62541/open62541/issues/1675)，I hope it will help you. @FishEatingCat
So,I think if we want to kill the problem,we need to add the type ,manually. And Now I have another problem as **createEvent failed. StatusCode BadNoMatch** 
Do you have any good suggestions? @basyskom-meerkoetter 
Looking forward for your reply. Thank you

Pro, 2019-02-25:
You first of all need to build the .h and .c file yourself. Since events is still experimental, it is not included by default in the releases.

Also, to build the events, you need to enable the full NS0 generation. This will create the BaseEventType in the server.

BaseEventType is an abstract type, this means that you can not directly create an event of this type. You need to create your own event type which is a subtype of BaseEventType, and use this type to emit events.

A working example is shown in https://github.com/open62541/open62541/blob/master/examples/tutorial_server_events.c

If these hints do not fix your issue, please create a new issue report with a detailed error description. You issue description should also include a MCVE (https://stackoverflow.com/help/mcve)

closed 2019-02-25 by Pro