open62541/open62541 #2389: BadTypeMismatch while creating matrix node

lporzio, 2019-01-24:
Hello!
I need help to set-up multidimensional arrays on a Server. 
My info:
- open62541 v 0.3 (latest one)
- CentOS 7

I started from the example _tutorial_server_variabletype.c_ and I'm trying to modify the code to add a matrix node to the Server. 

Here the steps I did:
- I changed the vAttr.valueRank to UA_VALUERANK_TWO_DIMENSIONS both in _addVariableType2DPoint_ and _addVariable_
- I increased arrayDims with:  UA_UInt32 arrayDims[2] = {2,2} in both functions
- I changed vtAttr.arrayDimensionsSize = 2 in both functions
- I changed UA_Double zero[4] = {0.0, 0.0, 0.0, 0.0}; and UA_Variant_setArray(&vtAttr.value, zero, 4, &UA_TYPES[UA_TYPES_DOUBLE]);

While it works for the 1-D array, with these changes I got the following errors:
```
[2019-01-24 10:36:37.296 (UTC+0000)] info/session	Connection 0 | SecureChannel 0 | Session 00000001-0000-0000-0000-000000000000 | WriteRequest returned status code BadTypeMismatch

[2019-01-24 10:36:37.296 (UTC+0000)] info/session	Connection 0 | SecureChannel 0 | Session 00000001-0000-0000-0000-000000000000 | AddNodes: Type-checking the variable node failed with error code BadTypeMismatch

[2019-01-24 10:36:37.296 (UTC+0000)] info/session	Connection 0 | SecureChannel 0 | Session 00000001-0000-0000-0000-000000000000 | AddNodes: Using attributes from the variable type failed with error code BadTypeMismatch
```

Here's the code. What is wrong?

```
/* This work is licensed under a Creative Commons CCZero 1.0 Universal License.
 * See http://creativecommons.org/publicdomain/zero/1.0/ for more information. */

/**
 * Working with Variable Types
 * ---------------------------
 *
 * Variable types have three functions:
 *
 * - Constrain the possible data type, value rank and array dimensions of the
 *   variables of that type. This allows interface code to be written against
 *   the generic type definition, so it is applicable for all instances.
 * - Provide a sensible default value
 * - Enable a semantic interpretation of the variable based on its type
 *
 * In the example of this tutorial, we represent a point in 2D space by an array
 * of double values. The following function adds the corresponding
 * VariableTypeNode to the hierarchy of variable types.
 */

#include <signal.h>
#include "open62541.h"

static UA_NodeId pointTypeId;

static void
addVariableType2DPoint(UA_Server *server) {
    UA_VariableTypeAttributes vtAttr = UA_VariableTypeAttributes_default;
    vtAttr.dataType = UA_TYPES[UA_TYPES_DOUBLE].typeId;
    //vtAttr.valueRank = UA_VALUERANK_ONE_DIMENSION;
    vtAttr.valueRank = UA_VALUERANK_TWO_DIMENSIONS;
    //UA_UInt32 arrayDims[1] = {2};
    UA_UInt32 arrayDims[2] = {2,2};
    vtAttr.arrayDimensions = arrayDims;
    //vtAttr.arrayDimensionsSize = 1;
    vtAttr.arrayDimensionsSize = 2;
    vtAttr.displayName = UA_LOCALIZEDTEXT("en-US", "2DPoint Type");

    /* a matching default value is required */
    //UA_Double zero[2] = {0.0, 0.0};
    UA_Double zero[4] = {0.0, 0.0, 0.0, 0.0};
    //UA_Variant_setArray(&vtAttr.value, zero, 2, &UA_TYPES[UA_TYPES_DOUBLE]);
    UA_Variant_setArray(&vtAttr.value, zero, 4, &UA_TYPES[UA_TYPES_DOUBLE]);

    UA_Server_addVariableTypeNode(server, UA_NODEID_NULL,
                                  UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE),
                                  UA_NODEID_NUMERIC(0, UA_NS0ID_HASSUBTYPE),
                                  UA_QUALIFIEDNAME(1, "2DPoint Type"), UA_NODEID_NULL,
                                  vtAttr, NULL, &pointTypeId);
}

/**
 * Now the new variable type for *2DPoint* can be referenced during the creation
 * of a new variable. If no value is given, the default from the variable type
 * is copied during instantiation.
 */

static UA_NodeId pointVariableId;

static void
addVariable(UA_Server *server) {
    /* Prepare the node attributes */
    UA_VariableAttributes vAttr = UA_VariableAttributes_default;
    vAttr.dataType = UA_TYPES[UA_TYPES_DOUBLE].typeId;
    //vtAttr.valueRank = UA_VALUERANK_ONE_DIMENSION;
    vAttr.valueRank = UA_VALUERANK_TWO_DIMENSIONS;
    //UA_UInt32 arrayDims[1] = {2};
    UA_UInt32 arrayDims[2] = {2,2};
    vAttr.arrayDimensions = arrayDims;
    //vtAttr.arrayDimensionsSize = 1;
    vAttr.arrayDimensionsSize = 2;
    vAttr.displayName = UA_LOCALIZEDTEXT("en-US", "2DPoint Variable");
    vAttr.accessLevel = UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE;
    /* vAttr.value is left empty, the server instantiates with the default value */

    /* Add the node */
    UA_Server_addVariableNode(server, UA_NODEID_NULL,
                              UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
                              UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT),
                              UA_QUALIFIEDNAME(1, "2DPoint Type"), pointTypeId,
                              vAttr, NULL, &pointVariableId);
}

/** It follows the main server code, making use of the above definitions. */

UA_Boolean running = true;
static void stopHandler(int sign) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "received ctrl-c");
    running = false;
}

int main(void) {
    signal(SIGINT, stopHandler);
    signal(SIGTERM, stopHandler);

    UA_ServerConfig *config = UA_ServerConfig_new_default();
    UA_Server *server = UA_Server_new(config);

    addVariableType2DPoint(server);
    addVariable(server);

    UA_StatusCode retval = UA_Server_run(server, &running);
    UA_Server_delete(server);
    UA_ServerConfig_delete(config);
    return (int)retval;
}
```

dwolfschlaeger, 2019-02-25:
Hello, 
I got the same issue, when trying to implement a new 2D-matrix datatype. Debugging my setup, I spotted that the arrayDimensions seem to read in the first value of UA_UInt32 arrayDims[2] = {2,2} only. It correctly identifies the address of the array, but considers only the very first element.

jpfr, 2019-08-01:
You have to add the matrix dimensions to the value.
Otherwise the variable expects a matrix. But the value is just an array.

A unit test and example is added in https://github.com/open62541/open62541/pull/2956.

commit a4cfd801 (2019-08-01): Server: Add an example for matrix variables. Fix #2389
closed 2019-08-01 by jpfr