open62541/open62541 #2392: client.cpp  Example

andDew15, 2019-01-26:
Hello everyone, 
I am newbie in OPC Communicationand I would like to ask you if you have some Cielnt.cpp Example ? In case of uabench not work. Is it possible to create it by opne64521.h lib ? 
Thanks

nicolasr75, 2019-01-28:
Have you tried the example from the home page down at the bottom?

https://github.com/open62541/open62541

That tutorial is a good start. Does that work for you?

To make it easy you should use a single file release from the Releases  page:
https://github.com/open62541/open62541/releases

Pro, 2019-01-30:
You can use any C Code inside a C++ file.
Have a look at the example here:
https://github.com/open62541/open62541/blob/master/examples/client.c

We currently do not have any c++ wrapper.

closed 2019-01-30 by Pro