open62541/open62541 #2394: Remaining memory leak with asynchronous connect

alajovic, 2019-01-28:
Issue #2295 has not been entirely fixed. I still get two blocks of memory marked "definitely lost" with the latest version of open62541. To reproduce, create the following files:

CMakeLists.txt

```
cmake_minimum_required(VERSION 3.0)

include(ExternalProject)

ExternalProject_Add(
  external_open62541
  GIT_REPOSITORY "https://github.com/open62541/open62541.git"
  GIT_TAG "fe01551f83df6f2b466a17e7695d78640ded6289"
  CMAKE_CACHE_DEFAULT_ARGS -DUA_ENABLE_AMALGAMATION:BOOL=ON
  INSTALL_COMMAND ""
)

ExternalProject_Get_Property(external_open62541 binary_dir)
set(OPEN62541_LIBRARY open62541)
set(OPEN62541_LIBRARY_PATH ${binary_dir}/bin/${CMAKE_FIND_LIBRARY_PREFIXES}${OPEN62541_LIBRARY}.a)
add_library(${OPEN62541_LIBRARY} UNKNOWN IMPORTED)
set_property(TARGET ${OPEN62541_LIBRARY} PROPERTY IMPORTED_LOCATION ${OPEN62541_LIBRARY_PATH})
set_property(TARGET ${OPEN62541_LIBRARY} PROPERTY INTERFACE_INCLUDE_DIRECTORIES ${binary_dir})
add_dependencies(${OPEN62541_LIBRARY} external_open62541)

add_executable(memory_leak_demo memory_leak_demo.c)
set_property(TARGET memory_leak_demo PROPERTY C_STANDARD 99)
target_link_libraries(memory_leak_demo ${OPEN62541_LIBRARY})
```

memory_leak_demo.c

```
#include <stdio.h>
#include <unistd.h>
#include <sys/wait.h>
#include <open62541.h>

static void
onConnect (UA_Client *client, void *userdata, UA_UInt32 requestId, void *status) {
    printf ("Async connect returned with status code %s\n",
            UA_StatusCode_name (*(UA_StatusCode *) status));
}

static void clientProcess() {
    UA_Client *client = UA_Client_new(UA_ClientConfig_default);
    const char* uri = "opc.tcp://localhost:4840";
    const int iterations = 20;

    for (unsigned int i = 0; i < iterations; i++) {
        UA_Client_connect_async(client, uri, onConnect, NULL);
        UA_Client_run_iterate(client, 0);
        usleep(250000);
    }

    UA_Client_delete(client);
}

static void serverProcess() {
    UA_ServerConfig* config = UA_ServerConfig_new_default();
    UA_Server* server = UA_Server_new(config);
    UA_Server_run_startup(server);

    // Run the server for 100 iterations = 5 seconds.
    for (int i = 0; i < 100; i++) {
        UA_Server_run_iterate(server, true);
    }

    UA_Server_run_shutdown(server);
    UA_Server_delete(server);
    UA_ServerConfig_delete(config);
}

int main() {
    if (fork()) {
        clientProcess();
        waitpid(-1, NULL, 0);
    } else {
        // Delay before starting the server process.
        sleep(2);
        serverProcess();
    }
    return 0;
}
```

Put the files side by side in a directory, and run

```
mkdir build
cd build
cmake ..
make
valgrind ./memory_leak_demo
```

Valgrind report:

```
==11313== 
==11313== HEAP SUMMARY:
==11313==     in use at exit: 0 bytes in 0 blocks
==11313==   total heap usage: 3,025 allocs, 3,025 frees, 1,462,154 bytes allocated
==11313== 
==11313== All heap blocks were freed -- no leaks are possible
==11313== 
==11313== For counts of detected and suppressed errors, rerun with: -v
==11313== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
==11312== 
==11312== HEAP SUMMARY:
==11312==     in use at exit: 288 bytes in 4 blocks
==11312==   total heap usage: 202 allocs, 198 frees, 947,775 bytes allocated
==11312== 
==11312== 288 (160 direct, 128 indirect) bytes in 2 blocks are definitely lost in loss record 2 of 2
==11312==    at 0x4C2DBB6: malloc (vg_replace_malloc.c:299)
==11312==    by 0x4707BE: UA_ClientConnectionTCP_init (open62541.c:51766)
==11312==    by 0x45CEF5: UA_Client_connect_async (open62541.c:43843)
==11312==    by 0x403FDE: clientProcess (in /home/alajovic/tmp/async-problem/build/memory_leak_demo)
==11312==    by 0x4040A3: main (in /home/alajovic/tmp/async-problem/build/memory_leak_demo)
==11312== 
==11312== LEAK SUMMARY:
==11312==    definitely lost: 160 bytes in 2 blocks
==11312==    indirectly lost: 128 bytes in 2 blocks
==11312==      possibly lost: 0 bytes in 0 blocks
==11312==    still reachable: 0 bytes in 0 blocks
==11312==         suppressed: 0 bytes in 0 blocks
==11312== 
==11312== For counts of detected and suppressed errors, rerun with: -v
==11312== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
```

The first part of the report (PID 11313) is for the server process – no leaks there – and the second (PID 11312) is for the client. 160 bytes in 2 blocks appear to be lost.

_Adapted from a post originally posted by @alajovic in https://github.com/open62541/open62541/issues/2295#issuecomment-454811714_

Infinity95, 2019-01-28:
Could you check if the problem still persists with the new networking code in #2271?

alajovic, 2019-01-28:
Yes, it does. Tested with https://github.com/Infinity95/open62541.git, branch feature/overhaul/network2, commit 18db252778568a5af0c6dd868a95f22ed824c44c. Valgrind output:

```
==4981== 
==4981== HEAP SUMMARY:
==4981==     in use at exit: 0 bytes in 0 blocks
==4981==   total heap usage: 3,034 allocs, 3,034 frees, 803,153 bytes allocated
==4981== 
==4981== All heap blocks were freed -- no leaks are possible
==4981== 
==4981== For counts of detected and suppressed errors, rerun with: -v
==4981== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
==4980== 
==4980== HEAP SUMMARY:
==4980==     in use at exit: 131,350 bytes in 5 blocks
==4980==   total heap usage: 133 allocs, 128 frees, 363,869 bytes allocated
==4980== 
==4980== 131,350 (144 direct, 131,206 indirect) bytes in 1 blocks are definitely lost in loss record 5 of 5
==4980==    at 0x4C2FB0F: malloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==4980==    by 0x183293: UA_TCP_DataSocket_allocate (open62541.c:52521)
==4980==    by 0x184187: UA_TCP_ClientDataSocket (open62541.c:52885)
==4980==    by 0x181A14: createDefaultClientSocket (open62541.c:51730)
==4980==    by 0x16FA32: UA_Client_connect_async (open62541.c:44792)
==4980==    by 0x11485F: clientProcess (in build/memory_leak_demo)
==4980==    by 0x114924: main (in build/memory_leak_demo)
==4980== 
==4980== LEAK SUMMARY:
==4980==    definitely lost: 144 bytes in 1 blocks
==4980==    indirectly lost: 131,206 bytes in 4 blocks
==4980==      possibly lost: 0 bytes in 0 blocks
==4980==    still reachable: 0 bytes in 0 blocks
==4980==         suppressed: 0 bytes in 0 blocks
==4980== 
==4980== For counts of detected and suppressed errors, rerun with: -v
==4980== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
```

Infinity95, 2019-01-28:
Should be fixed in #2271 now.
I can add a fix for the master branch if necessary.
Otherwise we can wait until #2271 is merged.

alajovic, 2019-02-18:
I have tested the code from https://github.com/Infinity95/open62541.git , commit 27f70dccd80dfd3660171cf37e7b7592266ae960. I had to modify my original example due to a change in the ClientConfig API:

```
--- memory_leak_demo.c.orig     2019-02-18 12:27:29.673510013 +0100
+++ memory_leak_demo.c  2019-02-18 12:28:04.637093535 +0100
@@ -10,7 +10,8 @@
 }
 
 static void clientProcess() {
-    UA_Client *client = UA_Client_new(UA_ClientConfig_default);
+    UA_Client *client = UA_Client_new();
+    UA_ClientConfig_setDefault(UA_Client_getConfig(client));
     const char* uri = "opc.tcp://localhost:4840";
     const int iterations = 20;
```

Can you please confirm that this modification is all that is needed to port the example to the new API? With the program modified in such a way, I get a bunch of warnings by valgrind about invalid read, such as:

```
==19874== Invalid read of size 8
==19874==    at 0x46F474: select_nm_shutdown (open62541.c:51424)
==19874==    by 0x431EDB: UA_Client_deleteMembers (open62541.c:35722)
==19874==    by 0x431F48: UA_Client_delete (open62541.c:35736)
==19874==    by 0x403D6A: clientProcess (in build/memory_leak_demo)
==19874==    by 0x403DFC: main (in build/memory_leak_demo)
==19874==  Address 0x5215840 is 0 bytes inside a block of size 24 free'd
==19874==    at 0x4C2ECB0: free (vg_replace_malloc.c:530)
==19874==    by 0x46F62F: select_nm_deleteMembers (open62541.c:51456)
==19874==    by 0x431D9F: UA_Client_deleteMembers (open62541.c:35689)
==19874==    by 0x431F48: UA_Client_delete (open62541.c:35736)
==19874==    by 0x403D6A: clientProcess (in build/memory_leak_demo)
==19874==    by 0x403DFC: main (in build/memory_leak_demo)
==19874==  Block was alloc'd at
==19874==    at 0x4C2DBB6: malloc (vg_replace_malloc.c:299)
==19874==    by 0x46F6FC: UA_SelectBasedNetworkManager (open62541.c:51477)
==19874==    by 0x46C096: configureClientNetworkManager_default (open62541.c:49988)
==19874==    by 0x43C635: UA_Client_run_iterate (open62541.c:40009)
==19874==    by 0x403D48: clientProcess (in build/memory_leak_demo)
==19874==    by 0x403DFC: main (in build/memory_leak_demo)
```

This is an excerpt only, I get multiple warnings similar to this one. I can post the full log if you need it.

Infinity95, 2019-02-18:
Since that commit i have done a lot of fixing regarding memory issues and changed the networking api some more.
I'd advise you to use at least commit 193485a12ff2ea3458c629f775e4a6232f5eff08.
If you use that commit, you need to create a new NetworkManager and register it with the client:
```
    /* Configure NetworkManager */
    UA_NetworkManager *networkManager;
    UA_StatusCode retval = UA_SelectBasedNetworkManager(UA_Log_Stdout, &networkManager);
    if(retval != UA_STATUSCODE_GOOD)
        return (int)retval;

    UA_Client_setNetworkManager(client, networkManager);
```

Remember to clean it up later:
```
    networkManager->shutdown(networkManager);
    networkManager->free(networkManager);
```

alajovic, 2019-02-21:
The problem is that I have a software product that uses the existing open62541 API, and I would like to stay with the master version until your networking overhaul is finished and tested. I cannot currently afford to modify everything in order to port it to the new networking API, especially because it appears to be work in progress.

I would much appreciate if you could also apply a fix on the master branch for the time being. Then we can continue on getting things working cleanly with the new networking code and API as well.

Infinity95, 2019-02-21:
@alajovic I ported the fix to master. Can you verify that this fixes you issue?
See #2483.

alajovic, 2019-02-21:
I took the current open62541 master (b41a64b31487d52e6f7d210e17a5d5e19886301c) and first merged in Infinity95/fix/pubsub (7869386353090a903c0eff73a7da349b2ed1e257) and then Infinity95/fix/2394 (c3e305f746b30c4a2eee5ab2edc31776213bf0b7). Linking my demo program with this, I get no memory leaks, but a bunch of blocks like this appears in the valgrind output:

```
==31357== Invalid read of size 8
==31357==    at 0x17D5A5: ClientNetworkLayerTCP_free (open62541.c:50444)
==31357==    by 0x14583F: UA_Client_connect_iterate (open62541.c:37437)
==31357==    by 0x14BD30: UA_Client_run_iterate (open62541.c:39606)
==31357==    by 0x113AC3: clientProcess (in build/memory_leak_demo)
==31357==    by 0x113B77: main (in build/memory_leak_demo)
==31357==  Address 0x5202860 is 48 bytes inside a block of size 80 free'd
==31357==    at 0x4C3011B: free (vg_replace_malloc.c:530)
==31357==    by 0x17D5C9: ClientNetworkLayerTCP_free (open62541.c:50446)
==31357==    by 0x14583F: UA_Client_connect_iterate (open62541.c:37437)
==31357==    by 0x14BD30: UA_Client_run_iterate (open62541.c:39606)
==31357==    by 0x113AC3: clientProcess (in build/memory_leak_demo)
==31357==    by 0x113B77: main (in build/memory_leak_demo)
==31357==  Block was alloc'd at
==31357==    at 0x4C2EEEF: malloc (vg_replace_malloc.c:299)
==31357==    by 0x17DBF5: UA_ClientConnectionTCP_init (open62541.c:50625)
==31357==    by 0x1459B6: UA_Client_connect_async (open62541.c:37465)
==31357==    by 0x113AB2: clientProcess (in build/memory_leak_demo)
==31357==    by 0x113B77: main (in build/memory_leak_demo)
==31357== 
```

So it is still not completely clean of memory problems.

alajovic, 2022-05-06:
Networking code refactor seems to have resolved this. Closing.