open62541/open62541 #2398: XML NodesetCompiler with custom datatypes not working

MampfHamster, 2019-01-29:
## Description

Hi,
I tried to use the server_testnodeset Example with my own .csv .bsd and .xml files but 
i got Several Compile errors. From my understanding, it is suficent to replace the three files in /examples/nodeset, or is there more to change so that the NodeSetCompiler can generate the .c and .h files for the server?
 I generated the files using the UA Modeler

.bsd [testtypes.txt](https://github.com/open62541/open62541/files/2807975/testtypes.txt)
.csv [testnodeset.txt](https://github.com/open62541/open62541/files/2808016/testnodeset.txt)
.xml [testnodeset.txt](https://github.com/open62541/open62541/files/2807981/testnodeset.txt)

But the resulting ua_types_testnodeset_generated.c does not contain any Types witch leads to errors like these:

Error	C2065	'UA_TestPointData': undeclared identifier	
Error	C2065	'UA_TYPES_TESTNODESET_TESTPOINTDATA': undeclared identifier

In Cmake i selected the full NS0 and the example files are generating .c .h files witch can be Compiled without Error

## Background Information / Reproduction Steps

## Checklist

Please provide the following information:

 - [X ] open62541 Version (release number or git tag):
f90a74c
 - [ ] Other OPC UA SDKs used (client or server): 
 - [X ] Operating system:
Windows 7 with MSVS 2015
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

MampfHamster, 2019-01-30:
To clarify the Orignial Files are working and result in a running Server which i can browse,
the ones created with UA Modeler result in Compile Errors

Pro, 2019-01-30:
Thanks for your issue report!

Interestingly there was a specific exclusion of types which contain 'Test' inside the name. This is now removed with #2403 

It's there since over 4 years and nobody ever used 'Test' in a type name, interesting :laughing: 

https://github.com/open62541/open62541/commit/b9bfc400d2d65a28db170aa470e1bff14d53a285

PR #2403 (2019-01-31): DatatypeGenerator: Also build types which have 'Test' in Name
Fixes #2398
commit a066163a (2019-01-31): DatatypeGenerator: Also build types which have 'Test' in Name

Fixes #2398
commit 7dc34299 (2019-03-08): DatatypeGenerator: Also build types which have 'Test' in Name

Fixes #2398
PR #2402 (2019-03-08): DatatypeGenerator: Also build types which have 'Test' in Name
Fixes #2398
closed 2019-01-30 by Pro