open62541/open62541 #2406: Compilation failure when compiling open62541 master branch in Visual Studio 2017 64bit

traversaro, 2019-01-31:
## Description
When the commit f85607bc7717771ff72e764b651a726ca92dbf5e is compiled on Windows with Visual Studio targeting 64 bit, the compilation is failling with the error: 
~~~
Severity	Code	Description	Project	File	Line	Suppression State
Warning	C4267	'function': conversion from 'size_t' to 'int', possible loss of data	open62541-plugins	C:\Users\STraversaro\src\open62541\arch\ua_network_tcp.c	125	
Error	C2220	warning treated as error - no 'object' file generated	open62541-plugins	C:\Users\STraversaro\src\open62541\arch\ua_network_tcp.c	125	
~~~
The main problem here is a warning on passing as third argument of `recv` a size_t variable.  On Windows the third argument of recv is indeed ìnt`, so the cast should be explicitly made. Otherwise, a warning is generated, and this warning generates an error, due to the `/CX` (treat warnings as errors)  flag present in the repo. 

A possible solution is to define a `OPC_CAST_THIRD_ARGUMENT_CAST` defined to `(int)` on Windows a to "" (empty string) in the other systems,  so you can use it as:
~~~
   ssize_t ret = UA_recv(connection->sockfd, (char*)&response->data[offset],
                          (OPC_CAST_THIRD_ARGUMENT_CAST) remaining, 0);
~~~ 

## Background Information / Reproduction Steps
Compile open62541 with the Visual Studio 2017 64-bit to reproduce the issue. 

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): f85607bc7717771ff72e764b651a726ca92dbf5e
 - [x] Other OPC UA SDKs used (client or server): none
 - [x] Operating system: Windows 10 
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached:  not relevant
 - [x] Wireshark network dump attached: not relevant
 - [x] Self-contained code example attached: not relevant 
 - [ ] Critical issue

traversaro, 2019-01-31:
I just noticed that in the Windows architecture file some macros are already defined to deal with the cast, so it is just necessary to follow the same style for `UA_recv`: https://github.com/open62541/open62541/blob/27e955528f9a96c5d8b5cc961716ff30815788f8/arch/win32/ua_architecture.h#L90 .

Pro, 2019-01-31:
Thanks for the fix!!

closed 2019-01-31 by Pro