open62541/open62541 #2410: Snapshot of all tags

rsoeldner, 2019-02-01:
Hi,

I'm looking for a way to capture a snapshot of **all availible** OPC tags, preferred all at once. 
Starting with the official 0.3 release I endet up with something like:
```
#include <stdio.h>
#include "open62541.h" 

int main(int argc, char *argv[]) {
    UA_Client *client = UA_Client_new(UA_ClientConfig_default);
    UA_StatusCode retval = UA_Client_connect(client, "opc.tcp://127.0.0.1:5000");
    if(retval != UA_STATUSCODE_GOOD) {
        UA_Client_delete(client);
        return (int)retval;
    }
   UA_BrowseRequest bReq;
   UA_BrowseRequest_init(&bReq);

   bReq.requestedMaxReferencesPerNode = 0; 
   bReq.nodesToBrowse = UA_BrowseDescription_new();
   bReq.nodesToBrowseSize = 1;
   bReq.nodesToBrowse[0].nodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
   bReq.nodesToBrowse[0].resultMask = UA_BROWSERESULTMASK_ALL;
   UA_BrowseResponse bResp = UA_Client_Service_browse(client, bReq);

   // loop over resultset bResp.resultsSize and bResp.results[i].referencesSize

  UA_BrowseRequest_deleteMembers(&bReq);
  UA_BrowseResponse_deleteMembers(&bResp);

  UA_Client_disconnect(client);
  UA_Client_delete(client);
}
````
To figure out the namespace and nodeid, but is there a direct way of retrieving all tags with values ?

Pro, 2019-02-01:
What exactly do you mean by tags?

Do you mean all the available nodes inside an opc ua server?

If yes, then have a look at this example:
https://github.com/open62541/open62541/blob/master/examples/client.c#L59

And start your browsing at the Root Folder:
```
bReq.nodesToBrowse[0].nodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ROOTFOLDER);
```

That's the best way to get a list of all nodes. You need to do this recursively, for all the children.

Please re-open the issue if this doesn't answer your question

closed 2019-02-01 by Pro

rsoeldner, 2019-02-09:
@Pro Thank you for the reply. Right now I'm able to browse the entire tree. Is there a way to get the full qualified name ?
```
void traverse(UA_Client *client, UA_NodeId nodeId) {
    UA_BrowseRequest bReq;
    UA_BrowseRequest_init(&bReq);
    bReq.requestedMaxReferencesPerNode = 0;
    bReq.nodesToBrowse = UA_BrowseDescription_new();
    bReq.nodesToBrowseSize = 1;

    UA_NodeId_copy(&nodeId, &bReq.nodesToBrowse[0].nodeId);
    bReq.nodesToBrowse[0].resultMask = UA_BROWSERESULTMASK_ALL; 

    UA_BrowseResponse bResp = UA_Client_Service_browse(client, bReq);

    for(size_t i = 0; i < bResp.resultsSize; ++i) {
        for(size_t j = 0; j < bResp.results[i].referencesSize; ++j) {
            UA_ReferenceDescription *ref = &(bResp.results[i].references[j]);
            if(ref->nodeClass == UA_NODECLASS_VARIABLE)
                    printf("%s, %s\n", ref->displayName.text.data, ref->browseName.name.data);
            else if(ref->nodeClass == UA_NODECLASS_OBJECT)
                    traverse(client, ref->nodeId.nodeId);
        }
    }

    UA_BrowseRequest_deleteMembers(&bReq);
    UA_BrowseResponse_deleteMembers(&bResp);
};

```