open62541/open62541 #2414: Server example don't pass the certification with CTT tool against basic DataAccess Server facet.

talsen-team, 2019-02-01:
## Description
I'm not sure if this is an issue, or maybe I just only need help and I'm using the sdk wrong. I created a very simple server, with just one node. Sample from the open62541 website. All is running well.

Then i downloaded the official "OPC Unified Architecture Compliance Test Tool (CTT)" tool from [opcfoundation](https://opcfoundation.org/)  website and tried to run the test against the DataAccess Server Profile facet (It's required for the project I'm currenlty working on, to pass the certification against it) tool screenshoot:

[image]

I get the following script result:
[image]
[image]

The connection to the server was created, and also a session was created, a node value was readed and so, everything is working.

But why are some server features / services missing (NotImplemented)? (called facets) like:

- Query,
- Attribute,
- MonitoredItems,
- Subscriptions (parts of the functions)

Do I need to enable them in the server ? Or implement something by my own?

I also tried the CTT against other OPC-UA SDKs, but sofar no SDK is passing the certification against the basic server DataAccess funtionality. (but the other SDKs have different services as test result: Implemented, or NotImplemented)

## Background Information / Reproduction Steps
Use opensdk to build sample server.
Get CTT tool, install and configure it to use the running server url.
Select DataAccess Server Facet profile and run test.

## Checklist

Please provide the following information:

 - [ x] open62541 Version (release number or git tag):  tag v0.3.0
 - [x ] Other OPC UA SDKs used (client or server): UniifiedAutomation C++, and HighPerformace SDK
 - [x ] Operating system: Ubuntu 18.04 Desktop LTS (minimal setup)
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [x ] Critical issue: No

jpfr, 2019-02-01:
The HistoryRead and HistoryUpdate services are developed on master and will be part of 0.4.
The DataAccess specific functionality is developed in this PR: https://github.com/open62541/open62541/pull/2238

So yes, we know DA is incomplete.
It is work-in-progress. You can track the features as they come in on Github.

teccheck, 2025-04-05:
The 1.4 family of releases is certified against standard server 2017. As there's been no recent activity, I'll close this issue

closed 2025-04-05 by teccheck