open62541/open62541 #2415: Export cmake target with a namespace ?

traversaro, 2019-02-01:
Currently, the CMake export targets of open62541 are installed as:
~~~cmake 
install(EXPORT open62541Targets
        FILE open62541Targets.cmake
        DESTINATION "${cmake_configfile_install}")
~~~
and then it is consumed as: 
~~~cmake
find_package(open62541)
add_executable(main main.cpp)
target_link_libraries(main open62541)
~~~
however, the CMake documentation recommends (see https://cmake.org/cmake/help/v3.13/manual/cmake-packages.7.html#creating-packages) to install exported targets specifying a `NAMESPACE`: 
> A NAMESPACE with double-colons is specified when exporting the targets for installation. This convention of double-colons gives CMake a hint that the name is an IMPORTED target when it is used by downstreams with the target_link_libraries() command. This way, CMake can issue a diagnostic if the package providing it has not yet been found. 

If a NAMESPACE was used, the code to consume the library will change to:
~~~cmake
find_package(open62541)
add_executable(main main.cpp)
target_link_libraries(main open62541::open62541)
~~~

To be honest, the difference is minimal: the main one is that if a user links a target whose name contains `::` but is not defined, CMake raises a fatal error at configuration time, while using a "simple" name, the target name is passed to the linker, moving the eventual error to the linking time. 

Given the minimal improvements due to the use of `NAMESPACE`, it probably does not make sense to disrupt existing users of the library. However, given that the 0.4 release will already bring some changes in the usage of the library, this is probably the right time to discuss if it is the case to migrate to use `open62541::open62541` as an exported target name or not.

ccvca, 2019-02-04:
> To be honest, the difference is minimal: the main one is that if a user links a target whose name contains `::` but is not defined, CMake raises a fatal error at configuration time, while using a "simple" name, the target name is passed to the linker, moving the eventual error to the linking time.
> 
That is a very useful error, because in most cases it wil fail to compile because some headers are missing (e.g. open62541.h) and you would start searching if you made a mistake including the headers...

Pro, 2019-02-04:
It sounds like a good idea to prefix with a namespace (See also https://stackoverflow.com/questions/48524359/naming-convention-for-components-and-namespaces-in-cmake)

Can you create a PR?

It will break the build for users on current master, but since it's dev version anyways, this small change is fine.

traversaro, 2019-02-04:
> Can you create a PR?

Ok!

closed 2019-02-15 by Pro