open62541/open62541 #2416: Local Discovery Server (LDS) does not work with dynamically assigned server ports

alex-lotz, 2019-02-02:
## Description

Local Discovery Server (LDS) does not support dynamically assigned port-numbers for a UA server using a server configuration as follows:

```
UA_ServerConfig *config = UA_ServerConfig_new_minimal(0, NULL);
```

The idea is to let the underlying OS decide about the next free port-number for the UA server. This also works, the UA server starts correctly and the OS assigns a dynamic port number. However, if the server registers itself to an LDS, it uses the wrong port number 0 (instead of the dynamically assigned port number). It seems that the UA server does not realize the changed port number which is fatal for LDS.

Possible solution: the UA server should use something like the Posix's **getsockname()**, at least for the special case of port-number 0, to detect the actually assigned port-number (which would be consistent with the Posix standard).

## Background Information / Reproduction Steps

I am using the **discovery** example within the open62541 library (see slightly modified example file attached). The idea is that multiple UA servers are started without manually assigning a static port number to each of them, but instead using the port-number 0 to get the next free port-number from the underlying OS. The UA server starts as expected and using the Ubuntu tool **sockstat** I can see that the underlying socket of the started UA server got a new (automatically assigned) port number. However, the UA server itself seems not to notice that it now has a different port-number than 0. Even worse, the UA server registers itself to the LDS with the wrong port number 0, see the terminal output of the LDS example below (this can also be seen by e.g. using the Ubuntu tool **avahi-discover**). If I now use the LDS to discover and to connect a UA client to my UA server, it fails as the client tries to connect to port 0.

```
> ./discovery_server_lds
[2019-02-02 11:44:37.737 (UTC+0100)] info/network	TCP network layer listening on opc.tcp://NB-ALotz:4840/
[2019-02-02 11:44:37.737 (UTC+0100)] info/server	Multicast DNS: add record for domain: LDS-NB-ALotz._opcua-tcp._tcp.local.
[2019-02-02 11:44:37.938 (UTC+0100)] info/server	Multicast DNS: found server: opc.tcp://NB-ALotz:4840
[2019-02-02 11:44:52.371 (UTC+0100)] info/network	Connection 6 | New connection over TCP from 127.0.0.1
[2019-02-02 11:44:52.371 (UTC+0100)] info/channel	Creating a new SecureChannel
[2019-02-02 11:44:52.371 (UTC+0100)] warn/securitypolicy	No PKI plugin set. Accepting all certificates
[2019-02-02 11:44:52.372 (UTC+0100)] info/channel	Connection 6 | SecureChannel 1 | Opened SecureChannel
[2019-02-02 11:44:52.372 (UTC+0100)] info/server	Multicast DNS: add record for domain: Sample Server-NB-ALotz._opcua-tcp._tcp.local.
[2019-02-02 11:44:52.523 (UTC+0100)] info/server	Multicast DNS: found server: opc.tcp://NB-ALotz:0
```
As can be seen in the last line of the LDS example output (above), the LDS discovers the new UA server with the wrong port-number 0.

## Checklist

Please provide the following information:

 - [X] open62541 Version (git commit): https://github.com/open62541/open62541/commit/fbc5763f60e9f7d29f6934f09a9bb116bccb7c0c
 - [X] Other OPC UA SDKs used (client or server): I am using the **discovery** example from the open62541 library, with a slightly modified "server_register.c" (see attached zip)
 - [X] Operating system: Ubuntu Linux 16.04 64bit
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [X] Self-contained code example attached: [server_register.c.zip](https://github.com/open62541/open62541/files/2824302/server_register.c.zip)
 - [ ] Critical issue

basyskom-meerkoetter, 2019-02-04:
I can confirm this issue.
I looked around a bit.
Adding the following snippet
```

    struct sockaddr_in returned_addr;
    bzero(&returned_addr, sizeof(returned_addr));
    socklen_t len = sizeof(returned_addr);
    getsockname(newsock, (struct sockaddr *)&returned_addr, &len);
    UA_UInt16 port = ntohs(returned_addr.sin_port);
    UA_LOG_WARNING(layer->logger, UA_LOGCATEGORY_NETWORK,
           "Listening on server port: %d", port);
```

to addServerSocket() in ua_network_tcp.c will print the dynamically assigned server ports.

Its unclear to me how to propagate this information to the right place so it will be used when registering with the LDS. Does it need to go into UA_ServerNetworkLayer->discoveryUrl?

Observation: these port numbers are different for the different interfaces of my laptop.

Pro, 2019-02-05:
I'll check that

alex-lotz, 2019-02-05:
Looking forward to see it in the master branch. Feel free to close this issue when you consider it closed.

Pro, 2019-05-03:
I created #2704 which allows to use port 0. This is an intermediate fix until the network refactoring PR is merged.

PR #2704 (2019-05-06): Stack: Add support for dynamic port assigning
Fixes #2416

Adds the possibility to use port `0` in the config, which means the OS will assign a random port.
commit 4a9bcf38 (2019-05-06): Stack: Add support for dynamic port assigning

Fixes #2416
closed 2019-05-06 by Pro