open62541/open62541 #2418: Error in CTT | Unsupported user token types

opcua-tsn-team-kalycito, 2019-02-04:
## Description
While running server_ctt application with secure channel (sign and basic128rsa15), the CTT test case "Session Services/Session Base/Err-009.js" fails.
Error Message : ActivateSession.Response.ResponseHeader.ServiceResult: BadIdentityTokenRejected (0x80210000); but Expected: BadIdentityTokenInvalid (0x80200000)

## Background Information / Reproduction Steps
This test case collects supported user token types from the server and test for unsupported types (In this case, X509 certificate and kerberos) for appropriate error response. The required error response is BadIdentityTokenInvalid (0x80200000) but it received BadIdentityTokenRejected (0x80210000).

- Is the X509 certificate user token policy implemented?
- Do we need to generate any certificate files specific to X509 before running server_ctt application?
- From the function Service_ActivateSession(arg1, ...) in the ua_services_session.c file, we see that the expected error response cannot be reached, as the X509 and kerberos are handled there.

**Steps to reproduce:**
1. Enable the test case "Session Services/Session Base/Err-009.js" in CTT
2. In CTT, configure Secure channel section as **sign** for MessageSecurityMode and  **Basic128Rsa15** for RequestedSecurityPolicyUri
3. Run the script

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): f97c524cb320236bcf42441a9d726f739258a121
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Debian GNU/Linux 9 (stretch)
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Regards
OPCUA-TSN-Team-Kalycito

ChHaa, 2019-02-05:
Not yet implemented afaik.

jpfr, 2019-02-08:
The authentication tokens are not handled by the core SDK but in the access control plugin.

https://github.com/open62541/open62541/blob/master/plugins/ua_accesscontrol_default.c#L104

@opcua-tsn-team-kalycito Can you provide a one-line PR that corrects the statuscode?

jpfr, 2019-02-08:
Update:
Ah ok. This fails even before since we try to match against the configured endpoints.
Still, one-line fix.

https://github.com/open62541/open62541/blob/master/src/server/ua_services_session.c#L320

PR #2452 (2019-02-14): [FIX] Return proper status code for unsupported UserToken types
 - Test case Err-009.js in Session services/Session base in CTT fails
   due to matching of userToken types against configured endpoints
 - The error status code to be returned is BadIdentityTokenInvalid
   (0x80200000) but obtained BadIdentityTokenRejected (0x80210000)
 - This fix will resolve the test and returns the expected status code
   (BadIdentityTokenInvalid - 0x80200000)

This pull request resolves #2418
closed 2019-02-14 by jpfr