open62541/open62541 #2424: CMake Policy CMP0058 Warning when using Ninja generator on Ubuntu 18.04

traversaro, 2019-02-04:
## Description
When configuring the project with CMake 3.10 on Ubuntu 18.04 using the Ninja generator, I get the following CMake warning, related to [CMake policy 0058](https://cmake.org/cmake/help/v3.10/policy/CMP0058.html):
~~~
CMake Warning (dev):
  Policy CMP0058 is not set: Ninja requires custom command byproducts to be
  explicit.  Run "cmake --help-policy CMP0058" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  This project specifies custom command DEPENDS on files in the build tree
  that are not specified as the OUTPUT or BYPRODUCTS of any
  add_custom_command or add_custom_target:

   doc_src/tutorials.rst

  For compatibility with versions of CMake that did not have the BYPRODUCTS
  option, CMake is generating phony rules for such files to convince 'ninja'
  to build.

  Project authors should add the missing BYPRODUCTS or OUTPUT options to the
  custom commands that produce these files.
This warning is for project developers.  Use -Wno-dev to suppress it.
~~~
I am not a frequent CMake+Ninja user and I did not read about Policy 0058 in detail, so I am not sure what is the best solution for this.  

## Background Information / Reproduction Steps
On a clean Ubuntu 18.04 system:
~~~
sudo apt install build-essential git cmake ninja-build
git clone https://github.com/open62541/open62541
cd open62541 
cmake -GNinja ..
~~~

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): https://github.com/open62541/open62541/commit/36fdd9779b1327192ace66ebdb0f4dc59bc77d0f
 - [x] Other OPC UA SDKs used (client or server): Not relevant
 - [x] Operating system: Ubuntu 18.04
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached: Not relevant
 - [x] Wireshark network dump attached: Not relevant
 - [x] Self-contained code example attached: Not relevant
 - [x] Critical issue: No

traversaro, 2019-02-04:
This PR was probably relevant: https://github.com/open62541/open62541/pull/1171 .

teccheck, 2025-04-05:
I've tested the current master state, and it seems to work using Ninja. I'll close this issue. Feel free to reopen if necessary

closed 2025-04-05 by teccheck

traversaro, 2025-04-06:
[`CMP0058`](https://cmake.org/cmake/help/latest/policy/CMP0058.html) was introduced in CMake 3.3, so setting as CMake minimum required to 3.13 as done by current master (https://github.com/open62541/open62541/blob/6ecfa9500761c681adb8cff9c5b92f5c76a38723/CMakeLists.txt#L1) implicitly sets it to `NEW`, and that what solved the problem.