open62541/open62541 #2429: compilation failed with tutorial_pubsub_subscribe sample

lakshmisivareddy, 2019-02-07:
Hi i am tying to cross compile open62541 to arm 

i am able to compile all the samples except tutorial_pubsub_subscribe

i am getting the bellow error 
i enabled UA_ENABLE_PUBSUB to ON 

Linking C executable ../bin/examples/tutorial_pubsub_subscribe
CMakeFiles/tutorial_pubsub_subscribe.dir/pubsub/tutorial_pubsub_subscribe.c.o: In function `subscriptionPollingCallback':
/home/opc-ua/open62541/examples/pubsub/tutorial_pubsub_subscribe.c:60: undefined reference to `UA_NetworkMessage_decodeBinary'
/home/opc-ua/open62541/examples/pubsub/tutorial_pubsub_subscribe.c:96: undefined reference to `UA_NetworkMessage_deleteMembers'
CMakeFiles/tutorial_pubsub_subscribe.dir/pubsub/tutorial_pubsub_subscribe.c.o: In function `run':
/home/opc-ua/open62541/examples/pubsub/tutorial_pubsub_subscribe.c:138: undefined reference to `UA_PubSubConnection_findConnectionbyId'
collect2: ld returned 1 exit status

lakshmisivareddy, 2019-02-28:
i am able to fix the issue by linking object files explicitly

closed 2019-02-28 by lakshmisivareddy

theresa95, 2019-03-05:
@lakshmisivareddy  I am also trying to compile the tutorial_pubsub_subscribe sample with gcc but it won't work. I managed to compile the tutorial_pubsub_publish sample, but everytime I try to compile the tutorial-pubsub_subscribe, I am getting errors that some include files are missing. It is the only example which does NOT include the open62541.c file too.

I tried to compile it like this:
gcc -std=c99 -o mySubscriber open62541.c tutorial_pubsub_subscribe.c

Please, I would be very grateful if you could lend me your help! I also started an issue on this topic a while ago but it was closed without me solving the actual problem. Please have a look if you need more details: https://github.com/open62541/open62541/issues/2306

Thank you so so much in advance!

lakshmisivareddy, 2019-03-05:
Hi @theresa95 
i explicitly added the object files which tutorial_pubsub_subscribe sample looking 
as shown below 

 add_example(tutorial_pubsub_subscribe pubsub/tutorial_pubsub_subscribe.c /home/opc-ua/latest/arm/ua_pubsub_networkmessage.c.o /home/opc-ua/latest/arm/ua_types_encoding_binary.c.o /home/opc-ua/latest/arm/ua_pubsub.c.o /home/opc-ua/latest/arm/ua_pubsub_manager.c.o /home/opc-ua/latest/arm/ua_timer.c.o)

theresa95, 2019-03-05:
HI @lakshmisivareddy , first of all, thank you so much for your quick reply! 
So you are building the tutorial_pubsub_subscribe sample with some object files? Where (or HOW) do you get these object files and how do you know which ones you need??

So if you are building this example with gcc, how do you build it?
Thank you so much!

theresa95, 2019-03-05:
And with which Building Options have you built the open62541 library??

lakshmisivareddy, 2019-03-06:
HI @theresa95 
when i am compiling the sample it is saying not able to find the symbol 
so i did 'grep' which object file having that symbol definition and linked it to the executable 
once it is compiled i run the sample it is working , interesting question here is 'libopen62541.so.0.4.0' showing it is having definition for all the symbols , i also don't the reason behind why sample  not able to find the symbols in .so file

theresa95, 2019-04-02:
hey @lakshmisivareddy , were you able to subscribe to value without specifying a network interface (instead of UA_STRING_NULL)?
`UA_NetworkAddressUrlDataType networkAddressUrl = {UA_STRING_NULL, UA_STRING("opc.udp://224.0.0.22:4840")};`

If yes, can you explain how to find the name of the network interface?

Is your arm microcontroller the subscriber? Who is publishing the values? Please let me know your setup for running the pubsub samples as I am trying to make my setup work ! Thank you in advance!