open62541/open62541 #2432: Authentication in a secured channel

davy7125, 2019-02-07:
Hello,

I need to initialize a connection with a username / password in a secured channel. I'm using for now these two lines:

`m_client = UA_Client_new(UA_ClientConfig_default);`
`UA_Client_connect_username(m_client, m_hostAddress, m_username, m_password);`

However I get the error code 0x80020000. I guess I need to specify somewhere that I need a secured channel SHA-256 but how can I do? I'm using version 3.0 of the open62541 library.

Thanks,
Davy

basyskom-meerkoetter, 2019-02-07:
Have you seen: https://github.com/open62541/open62541/blob/master/examples/encryption/client_encryption.c

davy7125, 2019-02-08:
Thank you for your fast answer. I read this code and here is why I didn't use it:

- I am using v0.3.0 and the function UA_Client_getConfig is not provided
- The software UA Expert can connect the machine and browse nodes simply with a username and a password. I'm thus expecting using just a username and password without the need of a certificate / private key. More information about UA Expert: I need to choose between Basic128Rsa15 and Basic256 since "None" gives the error (or warning) "BadSecurityModeInsufficient".

Should I use the version under development? Is there a generic certificate that use UA Expert?

I'm currently using this example:
https://github.com/open62541/open62541/blob/master/examples/client.c

I just replaced

```
UA_Client *client = UA_Client_new();
UA_ClientConfig_setDefault(UA_Client_getConfig(client));
```

by

`m_client = UA_Client_new(UA_ClientConfig_default);`

jpfr, 2019-02-08:
> I'm thus expecting using just a username and password without the need of a certificate / private key.

Some servers expect the password to be encrypted if it goes over an unencrypted channel.
So you get the full crypto infrastructure to do that.

UAExpert automatically generates a local certificate and uses that.
v.04 (master) just learned to encrypt only the password.
Expect more improvements in the coming time.

Potentially we can only use the remote certificate.
But atm we need the local certificate to instantiate the crypto context.

davy7125, 2019-02-11:
Ok I understand better now! I will try with v0.4
Thank you for the help

closed 2019-02-11 by davy7125