open62541/open62541 #2434: How to change the nodeId from numeric to String

Jmiciak67, 2019-02-08:
- [ ] open62541 Version (release number or git tag): last one
 - [ ] Other OPC UA SDKs used (client or server):  server
 - [ ] Operating system: w7 64 bit

Hello everyone, I am stuck in a project of OPC UA server. I am very new to this so I should miss something.
I follow several example to create my first server test. On this server I have:
- One main object called Object_1234
- in subobject I have Config and Result
- And in each of these object (config result) I have several variables.
Here is the three architecture:
[image]

The problem is that my variable have a numeric identifier and this is not usable to read or write a specific value
[image]

And when I am using the method "NewObject" it will create the same Object with another name, but also another identifier for the variable.
Concerning the subObject I have this:
[image]
[image]
[image]

The identifier is fixed thanks to the main Object name and the concatenate of ".config"

Is there some Example that show how to have a correct architecture tree like:
Object1
Object1.config
Object1.obj2
Object1.config.var1

Is there another way to access to my variable without using this numeric NodeId ? I have seen something like with the browse name maybe ?
I cannot send the code because it is confidential project, if you can just link some example on NodeId or browseName utilities.
Thanks all and best regards !

jpfr, 2019-02-08:
You can use

UA_Server_addNode_begin
UA_Server_addNode_finish

The child nodes are automatically added only during _finish.
If you add them manually in between (with the correct browsename), then the existing child nodes will be used.

https://github.com/open62541/open62541/blob/master/include/ua_server.h#L1178

Jmiciak67, 2019-02-08:
Thanks a lot jpfr ! your solution is working properly !
Best regards

closed 2019-02-08 by jpfr