open62541/open62541 #2436: Support for using open62541 via the CMake's FetchContent module

traversaro, 2019-02-08:
## Description
The CMake's Module [FetchContent](https://cmake.org/cmake/help/v3.11/module/FetchContent.html) (available since CMake 3.11) is an additional way with which CMake libraries can be integrated in a CMake-based build-system. With respect to the classical `find_package(...)` command, in which the CMake package is expected to be available in the system, with `FetchContent` the library's source code is downloaded during the CMake configure phase, and then the library is added to the project using the classical `add_subdirectory`, see for example: 
~~~cmake 
FetchContent_Declare(
  open62541
  GIT_REPOSITORY https://github.com/open62541/open62541
  GIT_TAG        ce4a4cbe02e712da446fed73be853051bac6b378
)

FetchContent_GetProperties(open62541)
if(NOT open62541_POPULATED)
  FetchContent_Populate(open62541)
  # Set open62541's options 
  set(UA_NAMESPACE_ZERO "FULL" CACHE STRING "" FORCE)
  add_subdirectory(${open62541_SOURCE_DIR} ${open62541_BINARY_DIR})
endif()

# ... 

target_link_libraries(main PRIVATE open62541)
~~~

This method is particularly convenient when cross-compiling using the [`CMAKE_TOOLCHAIN_FILE`](https://cmake.org/cmake/help/v3.13/variable/CMAKE_TOOLCHAIN_FILE.html) CMake option, as it ensures that open62541 is compiled with the exact same option of the rest of your project. With respect to creating an amalgamated `open62541.h`/`open62541.c` files and including them in the build, it has the advantage of keep using the same included headers (`ua_server.h`, `ua_client.h` ), and to simplify the process of updating `open62541` (you just need to change the `GIT_TAG` parameter. 

As of today, the `master` branch of `open62541` can't be used with the `FetchContent` module due to a few  issue in the build system. These issues can be easily solved without affecting in any way the other ways of using `open62541`, but I wanted to first open an issue about the usage of FetchContent, to understand if there is interest in this feature.

## Background Information / Reproduction Steps

See description. 

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): ce4a4cbe02e712da446fed73be853051bac6b378
 - [x] Other OPC UA SDKs used (client or server): Not relevant
 - [x] Operating system: Not relevant
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached: Not relevant
 - [x] Wireshark network dump attached: Not relevant
 - [x] Self-contained code example attached: Not relevant
 - [x] Critical issue: No

Pro, 2019-02-09:
Yes definitely!
Adding fetch content support would be really nice!

Could you provide a PR?

traversaro, 2019-02-09:
Ok!

traversaro, 2019-02-11:
First PR related to `FetchContent` support: https://github.com/open62541/open62541/pull/2447 .

traversaro, 2019-02-26:
Once https://github.com/open62541/open62541/pull/2447 gets merged,  I think the last thing to fix is that `open62541_TOOLS_DIR` and `open62541_NODESET_DIR` (used in the code-generating macros https://github.com/open62541/open62541/blob/master/tools/cmake/macros_public.cmake#L408 ) are not visible outside  the `open62541` subdirectory where they are defined: https://github.com/open62541/open62541/blob/73c9e78dfdef553f670be312a72ebb772992f883/CMakeLists.txt#L19 .

I guess a possible solution is to defined them either as  cache variables or as a GLOBAL PROPERTY, but to be honest I am not fully convinced by any of these solutions.

teccheck, 2025-05-16:
I tested support for `FetchContent` and it seems to work just fine. Thanks for the contribution @traversaro :)

closed 2025-05-16 by teccheck