open62541/open62541 #2438: Traversing entire tree and get all values

rsoeldner, 2019-02-09:
Hi,
I want to traverse the entire hirarchy down und print the values. The following code is what I ended with, but it becomes extreme slow for. Am I misusing the library ? 

Thanks in advance
```
void traverse(UA_Client *client, UA_NodeId nodeId) {
    UA_BrowseRequest bReq;
    UA_BrowseRequest_init(&bReq);
    bReq.requestedMaxReferencesPerNode = 0;
    bReq.nodesToBrowse = UA_BrowseDescription_new();
    bReq.nodesToBrowseSize = 1;

    UA_NodeId_copy(&nodeId, &bReq.nodesToBrowse[0].nodeId);
    bReq.nodesToBrowse[0].resultMask = UA_BROWSERESULTMASK_ALL; 

    UA_BrowseResponse bResp = UA_Client_Service_browse(client, bReq);

    for(size_t i = 0; i < bResp.resultsSize; ++i) {
        for(size_t j = 0; j < bResp.results[i].referencesSize; ++j) {
            UA_ReferenceDescription *ref = &(bResp.results[i].references[j]);
            if(ref->nodeClass == UA_NODECLASS_VARIABLE)
            {
                   UA_Variant* var = UA_Variant_new();
                   UA_Client_readValueAttribute(client, ref->nodeId.nodeId, var); 
                   UA_Variant_delete(var);
                UA_ReadResponse_deleteMembers(resp);
            }
            else if(ref->nodeClass == UA_NODECLASS_OBJECT)
                    traverse(client, ref->nodeId.nodeId);
        }
    }

    UA_BrowseRequest_deleteMembers(&bReq);
    UA_BrowseResponse_deleteMembers(&bResp);
};
```

jpfr, 2019-02-10:
Slow in what way?
- The server takes a long time for each service
- There are too many nodes

Note that you can speed this up by sending a read many variable values in the same ReadRequest.
This batch processing should be faster as it reduces the number of blocking network roundtrips.

rsoeldner, 2019-02-10:
@jpfr The server needs increasing more time for the response. Is my implementation reasonable?
I would assume for 4000 variables this should be done in less than 8 seconds (assuming 30ms response time).

jpfr, 2019-02-10:
On a local network it should be a lot less than 30ms.
Otherwise you'd have to wait 4000 / 33 = 121.2 seconds.

Which version of open62541 are you on?

jpfr, 2019-11-21:
Seems tobe resolved.

closed 2019-11-21 by jpfr