open62541/open62541 #2445: Node could not create a node with error code BadNodeAttributesInvalid

lakshmisivareddy, 2019-02-11:
Hi 
i am getting below error when i am running the 'tutorial_server_object'sample 

[2019-02-11 14:57:34.874 (UTC+0000)] info/session       Connection 0 | SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | AddNodes: Node could not create a node with error code BadNodeAttributesInvalid
[2019-02-11 14:57:34.875 (UTC+0000)] info/session       Connection 0 | SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | AddNodes: Node could not create a node with error code BadNodeAttributesInvalid

i am not able to see any fields in opcua client

jpfr, 2019-02-11:
I cannot reproduce on master.

```bash
jpfr@virtualbox:~/software/open62541/build$ ./bin/examples/tutorial_server_object
[2019-02-11 16:15:43.583 (UTC+0100)] info/userland      New pump created
[2019-02-11 16:15:43.583 (UTC+0100)] info/userland      New pump created
[2019-02-11 16:15:43.583 (UTC+0100)] info/network       TCP network layer listening on opc.tcp://virtualbox:4840/
^C[2019-02-11 16:15:45.964 (UTC+0100)] info/server      received ctrl-c
[2019-02-11 16:15:45.964 (UTC+0100)] warn/network       Socket select failed with Interrupted system call
[2019-02-11 16:15:45.964 (UTC+0100)] info/network       Shutting down the TCP network layer
```

Which version of open62541 are you using?

lakshmisivareddy, 2019-02-11:
i am using master only 
i cross compiled it for arm and testing on the board 
i am facing that error

jpfr, 2019-02-11:
It is difficult to help if we cannot replicate.
Can you increase the log level and/or find steps to replicate on a windows or linux system?

lakshmisivareddy, 2019-02-12:
Hi @jpfr 
thanks for the support i am able to fix the issue .

jpfr, 2019-02-13:
Great! Please reopen if required.

closed 2019-02-13 by jpfr

Darker7, 2019-08-23:
@lakshmisivareddy I'm facing the [same problem](https://github.com/open62541/open62541/issues/3027), how did you solve it?