open62541/open62541 #2449: client create subscription/monitored item issue

HamidMapna, 2019-02-13:
Hi. i am stuck in raising the datachange handler `(handler_changes)` while I change the value of monitored item. What i do is creating of subscription request and then adding a monitored item. But the callback handler doesn't raise when I change it's value. I've already defined `MyNodeTag` in server and I can observe it in UaExpert. When I change the value of `MyNodeTag` in UaExpert, i can catch the change and it's new value in `sampleCallbackWithValue(UA_Server *server, UA_MonitoredItem *monitoredItem,UA_DataValue *value)` method. But it goes through the true section of `if(sub){...}` in this method while i need it to go through the `else` section of condition where it calls the local callback which actually is my interested `handler_Changes`. I appreciate if someone enlighten me what I'm doing wrong.

`  

                        static void handler_Changes(UA_Client *client,UA_UInt32 subId,void *subContext,
		                UA_UInt32 monId,void *monContext,UA_DataValue *value)
                         {
	                        printf("answer changed\n");
	                        UA_Int32 readVal = 0;
	                        if(value->hasValue)
	                       {
		                    readVal = *(UA_Int32*) value->value.data;
		                    printf("value of monItem=%d\n",readVal);
	                       }
	                       else
		                    printf("Read value of monItem failed.\n");
                         }

               void clientCode()
                    {
                	   UA_UInt32 subId = 0;
			   UA_CreateSubscriptionRequest request;
                           UA_CreateSubscriptionRequest_init(&request);
			   request.requestedPublishingInterval = 500.0;
		           request.requestedLifetimeCount = 1000;
			   request.requestedMaxKeepAliveCount = 10;
			   request.maxNotificationsPerPublish = 0;
			  request.publishingEnabled = true;
			  request.priority = 0;
                          UA_CreateSubscriptionResponse 	response = 
                          UA_Client_Subscriptions_create(client,request,NULL,NULL,NULL);
			  subId = response.subscriptionId;
			  if(subId)
				printf("create subscription succeed,id=%u\n",subId);
				else
				printf("subscription failed.\n");
			   UA_MonitoredItemCreateRequest monRequest =
			   UA_MonitoredItemCreateRequest_default(UA_NODEID_STRING(0,"MyNodeTag"));
			  UA_MonitoredItemCreateResult monResponse =
                        UA_Client_MonitoredItems_createDataChange(client,subId,UA_TIMESTAMPSTORETURN_BOTH,monRequest,NULL,handler_Changes,NULL); 
                    if(monResponse.statusCode == UA_STATUSCODE_GOOD)
		        {
			printf("Monitoring id = %u\n",monResponse.monitoredItemId);
		        }
			else
			{
			printf("monResponse failed\n");
			}
		UA_Client_run_iterate(client,1000);        
                     }            
`

HamidMapna, 2019-02-13:
OK. Eventually i found the source of my problem. i had to create monitored item in server side using `UA_Server_createDataChangeMonitoredItem(...)` method. Of course this should be done after adding all interested variables.

StalderT, 2019-02-14:
You must call `UA_Client_run_iterate` permanently.

You can learn from the example that works : https://github.com/open62541/open62541/blob/master/examples/client_subscription_loop.c

jpfr, 2019-02-15:
StalderT found the root issue.
Hey lease reopen if required.

closed 2019-02-15 by jpfr