open62541/open62541 #2450: Build failure on master

shrmrf, 2019-02-13:
## Description
Building the latest master (`3a1e7fe`) with `gcc` version `5.4.0`, `clang` version `6.0.1` fails with the following error:

```
...
[ 59%] Linking C executable ../bin/examples/custom_datatype_client
[ 59%] Built target custom_datatype_client
[ 59%] Generating ../../src_generated/example_nodeids.h
make[2]: *** No rule to make target '../deps/ua-nodeset/Schema/Opc.Ua.NodeSet2.xml', needed by 'src_generated/ua_namespace_example.c'.  Stop.                                                                      
CMakeFiles/Makefile2:1663: recipe for target 'examples/nodeset/CMakeFiles/server_nodeset.dir/all' failed
make[1]: *** [examples/nodeset/CMakeFiles/server_nodeset.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2
```

## Background Information / Reproduction Steps
```bash
# inside the checked out directory
git checkout 3a1e7fec8a2a73ae93f7c15f08b5ee43adb010d8
mkdir build && cd build
cmake ..
make -DBUILD_SHARED_LIBS -DUA_BUILD_EXAMPLES -DUA_BUILD_TOOLS -DUA_BUILD_UNIT_TESTS
```
## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): 3a1e7fec8a2a73ae93f7c15f08b5ee43adb010d8
 - [X] Other OPC UA SDKs used (client or server): none, just built
 - [X] Operating system: Ubuntu 16.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [X] Self-contained code example attached: see above
 - [ ] Critical issue

Infinity95, 2019-02-13:
You will have to do
```
git submodule init && git submodule update
```
in order to build.

Maybe we can integrate these steps in the CMakeLists.txt file?

shrmrf, 2019-02-13:
@Infinity95 oh cool! thanks for that info. We should at least update the documentation. This would be really confusing for first-timers. I managed to build using `gcc` and `clang`. Thanks again!!

Pro, 2019-02-13:
This example should only be built if full NS0 is enabled. And then CMake would fail here, if you did not init the submodules:

```
if(NOT EXISTS "${UA_FILE_NS0}")
    message(FATAL_ERROR "File ${UA_FILE_NS0} not found. You probably need to initialize the git submodule for deps/ua-nodeset.")
endif()
```

We should change the cmake to exclude this example on non full NS0 build

Pro, 2019-02-25:
Fixed in #2475

closed 2019-02-25 by Pro