open62541/open62541 #2454: Constrain Variable Values

GLmontanari, 2019-02-14:
## Description
Dear all,
I'm moving my first steps in the OPC world and I was trying to understand if there is a way implemented in the open62541 library to constrain the range of possible values a Client can write in a Variable exposed from the Server.

## Background Information / Reproduction Steps
As an example, I created a Server based on the tutorial code which I've found in **Adding Variables to a Server**: https://open62541.org/doc/current/tutorial_server_variable.html.
I report the code below for sake of completeness:

```
#include <ua_server.h>
#include <ua_config_default.h>
#include <ua_log_stdout.h>

#include <signal.h>
#include <stdlib.h>

static void
addVariable(UA_Server *server) {
    /* Define the attribute of the myInteger variable node */
    UA_VariableAttributes attr = UA_VariableAttributes_default;
    UA_Int32 myInteger = 42;
    UA_Variant_setScalar(&attr.value, &myInteger, &UA_TYPES[UA_TYPES_INT32]);
    attr.description = UA_LOCALIZEDTEXT("en-US","the answer");
    attr.displayName = UA_LOCALIZEDTEXT("en-US","the answer");
    attr.dataType = UA_TYPES[UA_TYPES_INT32].typeId;
    attr.accessLevel = UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE;

    /* Add the variable node to the information model */
    UA_NodeId myIntegerNodeId = UA_NODEID_STRING(1, "the.answer");
    UA_QualifiedName myIntegerName = UA_QUALIFIEDNAME(1, "the answer");
    UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
    UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
    UA_Server_addVariableNode(server, myIntegerNodeId, parentNodeId,
                              parentReferenceNodeId, myIntegerName,
                              UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE), attr, NULL, NULL);
}
```

and

```
UA_Boolean running = true;
static void stopHandler(int sign) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "received ctrl-c");
    running = false;
}

int main(void) {
    signal(SIGINT, stopHandler);
    signal(SIGTERM, stopHandler);

    UA_ServerConfig *config = UA_ServerConfig_new_default();
    UA_Server *server = UA_Server_new(config);

    addVariable(server);

    UA_StatusCode retval = UA_Server_run(server, &running);
    UA_Server_delete(server);
    UA_ServerConfig_delete(config);
    return retval == UA_STATUSCODE_GOOD ? EXIT_SUCCESS : EXIT_FAILURE;
}
```

Now, I could connect to the server with a Client and type any value in the answer Node, as long as it is a int32, for instance -12345. Instead, I'd rather modify the Node properties so that its value can only range from -100 to 100.
How can I accomplish it?

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v0.3.0
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: WIndows 10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-02-14:
This is not possible with the standard OPC UA service.
The constraints for variables only apply to the datatype and the dimensionality.

You can however add a custom `DataSource` callback to the variable.
Then you land in you self-defined code where such constraints can be added.

jpfr, 2019-02-14:
Please reopen if required.

closed 2019-02-14 by jpfr

GLmontanari, 2019-02-14:
Ok, so... I was studying the part on DataSources in https://open62541.org/doc/current/tutorial_server_datasource.html . The example here deals with updating a DateTime variable, but I see that among the callback's parameters there is one named `UA_NumericRange`.
Can I use that for the purpose I am thinking?

Here is the example code:
```
static UA_StatusCode
readCurrentTime(UA_Server *server,
                const UA_NodeId *sessionId, void *sessionContext,
                const UA_NodeId *nodeId, void *nodeContext,
                UA_Boolean sourceTimeStamp, const UA_NumericRange *range,
                UA_DataValue *dataValue) {
    UA_DateTime now = UA_DateTime_now();
    UA_Variant_setScalarCopy(&dataValue->value, &now,
                             &UA_TYPES[UA_TYPES_DATETIME]);
    dataValue->hasValue = true;
    return UA_STATUSCODE_GOOD;
}
```

The callback is then used in the following way:

```
    UA_DataSource timeDataSource;
    timeDataSource.read = readCurrentTime;
    timeDataSource.write = writeCurrentTime;
    UA_Server_addDataSourceVariableNode(server, currentNodeId, parentNodeId,
                                        parentReferenceNodeId, currentName,
                                        variableTypeNodeId, attr,
                                        timeDataSource, NULL, NULL);
```

I think that I could use `UA_NumericRange` to do what you suggest, but how are its values set in first place? And how is it used from the callback since it is called with no arguments?

jpfr, 2019-02-14:
No, NumericRange is used to select ranges in an array of matrix.
The NumericRange is defined in the service call by the client.

GLmontanari, 2019-02-18:
Ok, I got it.