open62541/open62541 #2455: Bug : UA_Server_writeArrayDimensions

juangburgos, 2019-02-14:
## Description

Bug in `UA_Server_writeArrayDimensions`.

Call me crazy, but in [this line](https://github.com/open62541/open62541/blob/master/include/ua_server.h#L407), shouldn't it be `UA_ATTRIBUTEID_ARRAYDIMENSIONS` instead of `UA_ATTRIBUTEID_VALUE`?

Current code snippet below:

```c++
static UA_INLINE UA_StatusCode
UA_Server_writeArrayDimensions(UA_Server *server, const UA_NodeId nodeId,
                               const UA_Variant arrayDimensions) {
    return __UA_Server_write(server, &nodeId, UA_ATTRIBUTEID_VALUE,
                             &UA_TYPES[UA_TYPES_VARIANT], &arrayDimensions);
}
```

Expected code snippet:

```c++
static UA_INLINE UA_StatusCode
UA_Server_writeArrayDimensions(UA_Server *server, const UA_NodeId nodeId,
                               const UA_Variant arrayDimensions) {
    return __UA_Server_write(server, &nodeId, UA_ATTRIBUTEID_ARRAYDIMENSIONS,
                             &UA_TYPES[UA_TYPES_VARIANT], &arrayDimensions);
}
```

## Checklist

 - Branch : master
 - Critical issue

juangburgos, 2019-02-14:
I tried changing it to `UA_ATTRIBUTEID_ARRAYDIMENSIONS` myself, and checking if the `UA_Server_writeArrayDimensions` works as follows:

```c++
UA_Variant uaArrayDimensions;
UA_Variant_init(&uaArrayDimensions);
uaArrayDimensions.arrayDimensionsSize = 1;
UA_UInt32 arrayDims[1] = { 3 };
uaArrayDimensions.arrayDimensions = arrayDims;
st = UA_Server_writeArrayDimensions(m_qopcuaserver->m_server, m_nodeId, uaArrayDimensions);
```

But it fails with `UA_STATUSCODE_BADTYPEMISMATCH`, specifically in [this part of the code](https://github.com/open62541/open62541/blob/master/src/server/ua_services_attribute.c#L1323):

```c++
case UA_ATTRIBUTEID_ARRAYDIMENSIONS:
        CHECK_NODECLASS_WRITE(UA_NODECLASS_VARIABLE | UA_NODECLASS_VARIABLETYPE);
        CHECK_USERWRITEMASK(UA_WRITEMASK_ARRRAYDIMENSIONS);
        CHECK_DATATYPE_ARRAY(UINT32);
        GET_NODETYPE
        retval = writeArrayDimensionsAttribute(server, session, (UA_VariableNode*)node,
                                               type, wvalue->value.value.arrayLength,
                                               (UA_UInt32 *)wvalue->value.value.data);
        UA_Nodestore_release(server, (const UA_Node*)type);
        break;
```

Something is going wrong in `CHECK_DATATYPE_ARRAY(UINT32);`.

jpfr, 2019-02-15:
Your proposed change looks good!

Which value is currently set?
Probably the server prevents to set array dimensions that do not match the current value.

Error logging could be more explicit here...

juangburgos, 2019-02-19:
Setting the array dimensions ony works when creating the variable as follows:

```c++
#include "open62541.h"

int main(void) 
{
	// setup server
    UA_ServerConfig *config = UA_ServerConfig_new_default();
    UA_Server       *server = UA_Server_new(config);
	// add variable
	UA_NodeId pointVariableId;
    // node attributes
    UA_VariableAttributes vAttr = UA_VariableAttributes_default;
    vAttr.dataType              = UA_TYPES[UA_TYPES_DOUBLE].typeId;
    vAttr.valueRank             = UA_VALUERANK_ONE_DIMENSION;
    UA_UInt32 arrayDims[1]      = {2};
    vAttr.arrayDimensions       = arrayDims;
    vAttr.arrayDimensionsSize   = 1;
    vAttr.displayName           = UA_LOCALIZEDTEXT("en-US", "My Array Variable");
    vAttr.accessLevel           = UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE;
	// add node
    UA_Server_addVariableNode(server,
		                      UA_NODEID_NULL,
                              UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
                              UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT),
                              UA_QUALIFIEDNAME (1, "MyArray"), 
		                      UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE),
                              vAttr, 
		                      NULL, 
		                      &pointVariableId);
    
	UA_Variant newVAr;
	UA_Double new_data[2] = { 3.0, 4.0 };
	UA_Variant_setArray(&newVAr, new_data, 2, &UA_TYPES[UA_TYPES_DOUBLE]);
	UA_Server_writeValue(server, pointVariableId, newVAr);
	// run server
	UA_Boolean running = true;
    UA_StatusCode retval = UA_Server_run(server, &running);
    UA_Server_delete(server);
    UA_ServerConfig_delete(config);
    return (int)retval;
}
```

This is how it looks in UAExpert:

[image]

But it does **not work** when using the `UA_Server_writeArrayDimensions` method **after** the variable is created:

```c++
#include "open62541.h"

int main(void) 
{
	// setup server
    UA_ServerConfig *config = UA_ServerConfig_new_default();
    UA_Server       *server = UA_Server_new(config);
	// add variable
	UA_NodeId pointVariableId;
    // node attributes
    UA_VariableAttributes vAttr = UA_VariableAttributes_default;
	// add node
    UA_Server_addVariableNode(server,
		                      UA_NODEID_NULL,
                              UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
                              UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT),
                              UA_QUALIFIEDNAME (1, "MyArray"), 
		                      UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE),
                              vAttr, 
		                      NULL, 
		                      &pointVariableId);
    // [OK] set value
	UA_Variant newVAr;
	UA_Double new_data[2] = { 3.0, 4.0 };
	UA_Variant_setArray(&newVAr, new_data, 2, &UA_TYPES[UA_TYPES_DOUBLE]);
	UA_Server_writeValue(server, pointVariableId, newVAr);
	// [OK] set type
	UA_Server_writeDataType(server, pointVariableId, UA_NODEID_NUMERIC(0, UA_NS0ID_DOUBLE));
	// [OK] set rank
	UA_Server_writeValueRank(server, pointVariableId, UA_VALUERANK_ONE_DIMENSION);
	// [FAILS] set array dimensions
	// >> WriteRequest returned status code BadTypeMismatch
	UA_Variant uaArrayDimensions;
	uaArrayDimensions.arrayDimensions     = (UA_UInt32 *)(UA_Array_new(1, &UA_TYPES[UA_TYPES_UINT32]));
	uaArrayDimensions.arrayDimensions[0]  = 2;
	uaArrayDimensions.arrayDimensionsSize = (size_t)1;
	UA_Server_writeArrayDimensions(server, pointVariableId, uaArrayDimensions);
	// run server
	UA_Boolean running = true;
    UA_StatusCode retval = UA_Server_run(server, &running);
    UA_Server_delete(server);
    UA_ServerConfig_delete(config);
    return (int)retval;
}
```

It shows a `WriteRequest returned status code BadTypeMismatch` error. This is how it looks in UAExpert:

[image]

Setting the value, the data type and the rank works, why not the array dimensions? My feeling is that no one ever tested the `UA_Server_writeArrayDimensions` method.

As I mentioned, even after manually passing `UA_ATTRIBUTEID_ARRAYDIMENSIONS` to the `__UA_Server_write` method, is still **not working**.

jojastahl, 2019-02-20:
> I tried changing it to `UA_ATTRIBUTEID_ARRAYDIMENSIONS` myself, and checking if the `UA_Server_writeArrayDimensions` works as follows:
>
> ```c++
> UA_Variant uaArrayDimensions; [...]

You are setting up the variant in the wrong way. You need to pass the arrayDims as data, and eventually setting up arrayDimensions to describe your array in data.

But you should use `UA_Variant_setArray` to setup your variant and avoid manipulating internal variant state manually:

```c++
UA_Variant uaArrayDimensions;
UA_UInt32 arrayDims[1] = { 3 };
UA_Variant_setArray(&uaArrayDimensions, arrayDims, 1, &UA_TYPES[UA_TYPES_UINT32]);
st = UA_Server_writeArrayDimensions(m_qopcuaserver->m_server, m_nodeId, uaArrayDimensions);
```

Try this together with your fix proposed in your intial comment.

juangburgos, 2019-02-20:
Yes, I can confirm it works, using along the **proposed fix** 👍 

```c++
#include "open62541.h"

int main(void) 
{
	// setup server
	UA_ServerConfig *config = UA_ServerConfig_new_default();
	UA_Server       *server = UA_Server_new(config);
	// add variable
	UA_NodeId pointVariableId;
	// node attributes
	UA_VariableAttributes vAttr = UA_VariableAttributes_default;
	// add node
	UA_Server_addVariableNode(server,
		UA_NODEID_NULL,
		UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
		UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT),
		UA_QUALIFIEDNAME (1, "MyArray"), 
		UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE),
		vAttr, 
		NULL, 
		&pointVariableId);
	// [OK] set display name
	UA_Server_writeDisplayName(server, pointVariableId, UA_LOCALIZEDTEXT("en-US", "My Array Variable"));
	// [OK] set value
	UA_Variant newVAr;
	UA_Double new_data[2] = { 3.0, 4.0 };
	UA_Variant_setArray(&newVAr, new_data, 2, &UA_TYPES[UA_TYPES_DOUBLE]);
	UA_Server_writeValue(server, pointVariableId, newVAr);
	// [OK] set type
	UA_Server_writeDataType(server, pointVariableId, UA_NODEID_NUMERIC(0, UA_NS0ID_DOUBLE));
	// [OK] set rank
	UA_Server_writeValueRank(server, pointVariableId, UA_VALUERANK_ONE_DIMENSION);
	// [OK] set array dimensions
	UA_Variant uaArrayDimensions;
	UA_UInt32 arrayDims[1] = { 2 };
	UA_Variant_setArray(&uaArrayDimensions, arrayDims, 1, &UA_TYPES[UA_TYPES_UINT32]);
	UA_Server_writeArrayDimensions(server, pointVariableId, uaArrayDimensions);
	// run server
	UA_Boolean running = true;
	UA_StatusCode retval = UA_Server_run(server, &running);
	UA_Server_delete(server);
	UA_ServerConfig_delete(config);
	return (int)retval;
}
```

I am closing this issue, I suppose someone in charge of the repo **will apply** the proposed solution?

Thanks.

PR #2478 (2019-02-21): Fix UA_Server_writeArrayDimensions
Fix #2455
closed 2019-02-20 by juangburgos