open62541/open62541 #2456: Error 0x8002000 may be too generic for certificates?

davy7125, 2019-02-15:
Hello,

I'm trying to connect a machine with OpcUa and this machine requires a certificate to secure the authentication process. So I created one with the application "opc.ua.certificategenerator.exe" in the simplest way:

`.\opc.ua.certificategenerator.exe -cmd issue -sp . -an MyApp`

I got a .der file for the certificate and a .pfx file for the private key.

Following this example: https://github.com/open62541/open62541/blob/master/examples/encryption/client_encryption.c
I configured the connection by providing the 2 files (using a .pfx extension instead of a .der extension mentioned in the example) and I provided no trust list since I don't know how to generate it.

When I try to connect I get the error 0x80020000 which is "An internal error occurred as a result of a programming or configuration error". Is it possible to be more specific here:
* maybe the certificate is not valid,
* maybe the format of the private key is not recognized (I used a .pfx instead of .der),
* maybe I should have provided a trust list (which seems to be a .ctl extension instead of .crl as mentioned in the example),
* maybe the server rejected it for some reason?

Regards,
Davy

schroeder-, 2019-02-15:
Mbedtls doesn't support .pfx, either use ".der" like in the examples or use ".pem" and add "\0" at the end of the string.

davy7125, 2019-02-15:
Thank you I adapted the command to
`.\opc.ua.certificategenerator.exe -cmd issue -sp . -an MyApp -pem true -ks 2048`
(I added the key length 2048 according to this page https://github.com/open62541/open62541/issues/1704)

I have however the same error when using the function "UA_Client_connect_username" => 0x80020000.

So I am checking the functions I am using:
```
m_client = UA_Client_new();
UA_ClientConfig *cc = UA_Client_getConfig(m_client);
UA_ClientConfig_setDefaultEncryption(cc, UA_BYTESTRING(m_certificate), UA_BYTESTRING(m_privateKey), NULL, 0, NULL, 0);
cc->securityMode = UA_MESSAGESECURITYMODE_SIGNANDENCRYPT;
retVal = UA_Client_connect_username(m_client, m_hostAddress, m_username, m_password;
```

I am wandering where I should mention that I need the endpoint with th eencryption Basic128Rsa15? I discovered that the function "UA_SecurityPolicy_Basic128Rsa15" exists but I don't know if I should use it. Or should I browse all endpoints, knowing that I need to use the second one (according to UaExpert)?

I also found a that doesn't exist anymore in the repo: examples/encryption/client_basic128rsa15.c. Maybe I should follow this example?

schroeder-, 2019-02-15:
Try:
`cc->securityPolicyUri = UA_STRING_ALLOC("http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15");`

davy7125, 2019-02-15:
Unfortunately this is the same. It's like I'm in the same situation than in the ticket I mentioned (https://github.com/open62541/open62541/issues/1704)

[image]

It's been fixed recently in master (I have the latest sources) so maybe everything has not been tested yet?
As additional information open62541 is successful when I try to connect with no certificate and with no authentication so my variables should be OK. But in that case I have no access (no namespace). My login / password are OK according to UaExpert.

schroeder-, 2019-02-15:
Are there any log messages generated? Otherwise set UA_LOGLEVEL to 300 or less. 
Connecting with encrypted password is working with Basic256Sha256 is me on master.

davy7125, 2019-02-15:
Good suggestion. I'll debug this on Monday and give you the feedback.

Infinity95, 2019-02-15:
It could be that the server refuses the connection because the client doesn't have the ApplicationDescription configured.
You could try setting this in the client config.
(I would start with setting the applicationName to the name you set in the certificate)

jpfr, 2019-02-15:
Master has seen quite a few improvements.

The client automatically selects a possible combination of encryption and login mechanism.

This choice is explicitly logged.
The server itself should be able to output detailed logs.

davy7125, 2019-02-18:
Here are the logs
```
2019-02-18 05:04:59,693 INFO: Success in creating a client
2019-02-18 05:04:59,693 INFO: Configuring a secured connection with the certificate
2019-02-18 05:04:59,693 INFO: Try to connect to 'opc.tcp://192.168.XX.XX:4840' with username = 'OpcUaClient' and password = 'XX'
2019-02-18 05:04:59,693 INFO: [Open62541, client] Connecting to endpoint opc.tcp://192.168.XX.XX:4840
2019-02-18 05:04:59,693 WARN: [Open62541, security policy] No PKI plugin set. Accepting all certificates
2019-02-18 05:04:59,709 INFO: [Open62541, client] TCP connection established
2019-02-18 05:04:59,709 INFO: [Open62541, client] Opened SecureChannel with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
2019-02-18 05:04:59,709 INFO: [Open62541, client] Endpoint and UserTokenPolicy unconfigured, perform GetEndpoints
2019-02-18 05:04:59,709 ERROR: [Open62541, client] No suitable endpoint found
2019-02-18 05:04:59,709 ERROR: LibOpen62541: connection to opc.tcp://192.168.XX.XX:4840 => error 0x80020000
```

In my test I didn't adapt the application name yet as suggested by @Infinity95 but I think I have an issue prior to the application name test. What is weird is that I specified a security policy and it's not taken into account.

```
cc->securityPolicyUri = UA_STRING_ALLOC("http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15");
```

I adapted the wiki for specifying a new logger: https://github.com/open62541/open62541/wiki/Using-open62541-from-C%23

jpfr, 2019-02-20:
The SecurityPolicyUri in the config should be taken into account.
But the client first opens an unsecured connection to discover the endpoints.
Then an endpoint is selected (with the configured SecurityPolicyUri) and a new SecureChannel is opened.

Can you see which endpoints are returned by `GetEndpoints`?
E.g. in UAExpert or Wireshark.

Otherwise you can track the endpoint selection with a debugger here:
https://github.com/open62541/open62541/blob/e2df239750f2f8cbcbd762c0540e6a95e6b82724/src/client/ua_client_connect.c#L533

davy7125, 2019-02-28:
I added logs (in my local open62541 repository) and it sounds like the security policy to use is not configured correctly. Thanks to @schroeder- I have been able to correctly set the variable **client->config.securityPolicyUri** but the second endpoint is rejected because of the variables **client->config.securityPoliciesSize** and **client->config.securityPolicies**, used in the function "getSecurityPolicy". How can I configure them?

Here is the code and logs for more details:
```
// Creation of a client
UA_Client * client = UA_Client_new();

// Early configuration
UA_ClientConfig *cc = UA_Client_getConfig(client);
UA_ClientConfig_setDefault(cc);
cc->logger.log = &LOG_LEVEL_1;
cc->clientDescription.applicationName.text = UA_STRING_ALLOC("MySoftware");
cc->clientDescription.applicationType = UA_APPLICATIONTYPE_CLIENT;

// List endpoints, use the function "UA_Client_getEndpoints"
this->ListEndpoints(client);

// Configuration
cc->securityMode = UA_MESSAGESECURITYMODE_SIGNANDENCRYPT;
UA_ClientConfig_setDefaultEncryption(cc, UA_BYTESTRING(m_certificate), UA_BYTESTRING(m_privateKey), NULL, 0, NULL, 0);
cc->securityPolicyUri = UA_STRING_ALLOC("http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15");
cc->logger.log = &LOG_LEVEL_1;

// Connection
UA_StatusCode retVal = UA_Client_connect_username(client, m_hostAddress, m_username, m_password);
```

And the associated logs:
```
2019-02-28 04:46:20,051 INFO: LibOpen62541: Success in creating a client
2019-02-28 04:46:20,051 INFO: LibOpen62541: browsing endpoints
2019-02-28 04:46:20,051 WARN: [Open62541, security policy] No PKI plugin set. Accepting all certificates
2019-02-28 04:46:20,067 INFO: [Open62541, client] TCP connection established
2019-02-28 04:46:20,067 INFO: [Open62541, client] Opened SecureChannel with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
2019-02-28 04:46:20,067 INFO: LibOpen62541: found 3 endpoint(s)
2019-02-28 04:46:20,067 INFO: LibOpen62541: URL of endpoint 0 is opc.tcp://192.168.55.7:4840, security mode = 'none (1)', security policy = 'http://opcfoundation.org/UA/SecurityPolicy#None', security level = '0', binary transport = 'http://opcfoundation.org/UA-Profile/Transport/uatcp-uasc-uabinary'
2019-02-28 04:46:20,067 INFO: LibOpen62541: URL of endpoint 1 is opc.tcp://192.168.55.7:4840, security mode = 'sign and encrypt (3)', security policy = 'http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15', security level = '0', binary transport = 'http://opcfoundation.org/UA-Profile/Transport/uatcp-uasc-uabinary'
2019-02-28 04:46:20,067 INFO: LibOpen62541: URL of endpoint 2 is opc.tcp://192.168.55.7:4840, security mode = 'sign and encrypt (3)', security policy = 'http://opcfoundation.org/UA/SecurityPolicy#Basic256', security level = '0', binary transport = 'http://opcfoundation.org/UA-Profile/Transport/uatcp-uasc-uabinary'
2019-02-28 04:46:20,067 INFO: LibOpen62541: configuring a secured connection with the certificate
2019-02-28 04:46:20,067 INFO: LibOpen62541: try to connect to 'opc.tcp://192.168.55.7:4840' with username = 'OpcUaClient' and password = 'XX'
2019-02-28 04:46:20,067 INFO: [Open62541, client] Connecting to endpoint opc.tcp://192.168.55.7:4840
2019-02-28 04:46:20,067 WARN: [Open62541, security policy] No PKI plugin set. Accepting all certificates
2019-02-28 04:46:20,067 INFO: [Open62541, client] TCP connection established
2019-02-28 04:46:20,067 INFO: [Open62541, client] Opened SecureChannel with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
2019-02-28 04:46:20,067 INFO: [Open62541, client] Endpoint and UserTokenPolicy unconfigured, perform GetEndpoints
2019-02-28 05:45:08,598 INFO: [Open62541, client] Found 3 endpoints
2019-02-28 05:45:08,598 INFO: [Open62541, client] Rejecting endpoint 0: security mode doesn't match
/!\ 2019-02-28 05:45:08,598 INFO: [Open62541, client] Rejecting endpoint 1: security policy not available /!\
2019-02-28 05:45:08,598 INFO: [Open62541, client] Rejecting endpoint 2: security policy doesn't match
2019-02-28 04:46:20,083 ERROR: [Open62541, client] No suitable endpoint found
2019-02-28 04:46:20,083 ERROR: LibOpen62541: connection to opc.tcp://192.168.55.7:4840 => error 0x80020000
```

davy7125, 2019-02-28:
I have been able to go further when using 
```
// Configuration
cc->securityMode = UA_MESSAGESECURITYMODE_SIGNANDENCRYPT;
UA_CertificateVerification_AcceptAll(&cc->certificateVerification);
cc->securityPolicyUri = UA_STRING_ALLOC("http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15");

// Populate security policies 
cc->securityPoliciesSize = 1;
cc->securityPolicies = new UA_SecurityPolicy[1];
retVal = UA_SecurityPolicy_Basic128Rsa15(
  &cc->securityPolicies[0],
  &cc->certificateVerification,
  UA_BYTESTRING(m_certificate),
  UA_BYTESTRING(m_privateKey),
  &cc->logger);
```

Instead of
```
cc->securityMode = UA_MESSAGESECURITYMODE_SIGNANDENCRYPT;
UA_ClientConfig_setDefaultEncryption(cc, UA_BYTESTRING(m_certificate), UA_BYTESTRING(m_privateKey), NULL, 0, NULL, 0);
cc->securityPolicyUri = UA_STRING_ALLOC("http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15");
```

I'm thus closing this issue. It could be fine to attach an example somewhere.

closed 2019-02-28 by davy7125