open62541/open62541 #2460: Please HELP!! How can I make my own single .c/.h file für STM32 Keil?

VladBraz125, 2019-02-18:
Hello experts,
I never used CMake, maybe that's why I have problems with make my own single .c/.h file.
I have already tried with example of Android from Pro, and main thread. I have no success.
My system is STM32F7 board, compiler Keil µVision5, freeRTOS and lwip. I developing under Windows 7.
Please help my!!!

## Checklist

Please provide the following information:
 - [x] open62541 Version (actual):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x ] Operating system: Windows 7 Pro 
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

adechatre, 2019-02-18:
Hi Vlad,
You can get .h/.c file from here : https://github.com/open62541/open62541/releases
Or follow : https://open62541.org/doc/current/building.html
Augustin

VladBraz125, 2019-02-18:
Thanks, but with files .h/.c on compiling i have a error # include <unistd.h> //access.
And  the building.html describes a process for Linux, OpenBSD, OSX, Windows.
If i create .h/.c files and compile for Visual Studio 2013 and try use it in Keil, I recieve compile errors.

adechatre, 2019-02-18:
Can we get the logs from Visual studio?

VladBraz125, 2019-02-18:
this is my command line
cmake -DUA_ENABLE_AMALGAMATION=true -DUA_ARCHITECTURE=freertosLWIP -DCMAKE_C_COMPILER="D:\Program Files (x86)\GNU Tools ARM Embedded\8 2018-q4-major\bin\arm-none-eabi-gcc" -DCMAKE_CXX_COMPILER="D:\Program Files (x86)\GNU Tools ARM Embedded\8 2018-q4-major\bin\arm-none-eabi-g++" -DCMAKE_FIND_ROOT_PATH="D:\Program Files (x86)\GNU Tools ARM Embedded\8 2018-q4-major\arm-none-eabi"

and this is a log from Visual studio
[CompileLog.txt](https://github.com/open62541/open62541/files/2875323/CompileLog.txt)

VladBraz125, 2019-02-18:
Fehler	1	error C1083: Datei (Include) kann nicht geöffnet werden: "unistd.h": No such file or directory	d:\downloads\opc ua tools\open62541\open62541\build\open62541.h	180	1	open62541-object
Fehler	2	error LNK1181: Eingabedatei "D:\Downloads\OPC UA Tools\open62541\open62541\build\open62541-object.dir\Debug\open62541.obj" kann nicht ge÷ffnet werden.	D:\Downloads\OPC UA Tools\open62541\open62541\build\LINK	open62541 (open62541\lib\open62541)

schroeder-, 2019-02-18:
have you tried
`cmake --build .`

Pro, 2019-02-18:
In your CMake you are setting the architecture to freertosLWIP but you try to build on Visual Studio on Windows. That was never tested and will most definitely not work. Either use Linux, or select the correct arch if you just want to build for your windows system

Pro, 2019-02-18:
Also as a general hint: please avoid starting your issue title with "Please help". Its better if you think about a good description, that will definitely attract more people to read your issue.

VladBraz125, 2019-02-18:
2schroeder: cmake --build give a same error as Visual Studio.
2Pro: Sorry, I tried a Week to bind open62541 on my project, where I use freeRTOS and lwip. How I can create make my own single .c/.h file für STM32 Keil. If you can please help me.

jpfr, 2019-02-18:
If you use a cmake GUI (like ccmake), you can see the build options and the advanced build options.

It shouls be enough to set

UA_ARCHITECTURE                  freertosLWIP
UA_ENABLE_AMALGAMATION           ON

I cannot test freeRTOS on my box.
If this does not work, please report. And we will fix it.

schroeder-, 2019-02-19:
You cant choose any Visual Studio because this will set the compiler to your Visual Studio C++ Compiler for Windows.

VladBraz125, 2019-02-19:
regardless of what settings I take for ARM with or without  freertosLWIP,  I have a compiler error "unistd.h": No such file or directory

adechatre, 2019-02-19:
Can you try to compile using mingGW?

Pro, 2019-02-25:
I would first suggest that you get a virtual machine with Ubuntu or some linux system where you try to build the stack with Posix. When that is working, switch to your specific architecture and make sure all the dependencies are installed for STM32.

For your custom architectures you can not use a pre-created .c/.h file, you need to create it yourself using the correct CMake settings.

We tested compiling for STM32 only on Linux, not on windows.

I guess since you are on Windows, the architecture include directories are not correctly set or found.
For your first tests disable amalgamation and make sure that you can build it with that settings. Then post the exact place where the `#include "unistd.h"` is listed.

Another thing you can try:
Just remove this line here:
https://github.com/open62541/open62541/blob/b2f29fa33d4ba8db65e23fe2fc810954db339dc3/arch/common/ua_freeRTOS.h#L17

rudoq32, 2019-03-05:
Hi @VladBraz125, did you succeed?  I am wondering, because I am also trying to bring it on nucleo f767zi.

Pro, 2019-03-05:
@rudoq32 please have a look at #2511 which includes a more detailed how-to and some more fixes.

Pro, 2019-03-08:
Please continue any discussion on STM32 and freeRTOS in #2511

PR #2511 (2019-04-04): WIP: Cleanup of freeRTOS and LwIP
I want to use this PR to fix all issues related to freeRTOS and LwIP: 
fixes #1859 
fixes #1864 
fixes #2227 
fixes #2460. 

Users working with freeRTOS that confirmed the PR works:

- [X] @embbo (https://github.com/open62541/open62541/pull/2511#issuecomment-475769413)
- [ ] @nidhal-mars (not active in the PR)
- [ ] @FishEatingCat (not active in the PR)
- [X] @VladBraz125 (https://github.com/open62541/open62541/pull/2511#issuecomment-479844345)
- [X] @rudoq32 (https://github.com/open62541/open62541/pull/2511#issuecomment-474811430)
- [ ] @sammer-h (not active  [... cut]
closed 2019-03-08 by Pro