open62541/open62541 #2462: how to compare to different UA_VARIANT (old und new)!

bnadem, 2019-02-18:
I want to store old value from Subscription in old variabel vom data typ UA_VARIANT, when i becommen new data from Subscription, i want ot compare new value (UA_VARIANT) with old value (UA_VARIABT),  becaus i becomme same data from Subscription!

## Description

## Background Information / Reproduction Steps

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

basyskom-meerkoetter, 2019-02-18:
You describe what you want to achieve but you don't describe what your specific problem is.
Can you please outline the steps taken so far and what problem you have encountered?

For working with variants in general have a look here: https://github.com/open62541/open62541/blob/master/examples/tutorial_datatypes.c
It can be also interesting to have a look at the unit tests covering UA_Variant. See here: https://github.com/open62541/open62541/blob/master/tests/check_types_builtin.c

bnadem, 2019-02-18:
When subscription for node enable, i becom same value Every subcription
time interval

juangburgos, 2019-02-19:
I think he is looking for a method like looks as follows:

```c++
static UA_INLINE UA_Boolean
UA_Variant_isEqual(const UA_Variant *v1, const UA_Variant *v2) {
    return /*TODO*/;
}
```

I also could not find such method in the code.

jpfr, 2019-02-20:
`UA_Variant_isEqual`

The simplest way to achieve this is to encode the value in the binary format and strcmp the result.
The same is done to for value changes via a MonitoredItem.

Pro, 2019-02-25:
Maybe we can wrap this code into a separate UA_Variant_isEqual and provide it to the public API?

https://github.com/open62541/open62541/blob/81958f67cac7d57ca19be91d9b1c0469b2e6b29f/src/server/ua_subscription_datachange.c#L235-L267

o6-nbeier, 2026-10-20:
Thank you for taking the time to open this issue and describe it so clearly.

The problem has since been fixed on the maintained branches (see d19c958), so we are closing it as resolved.

closed 2026-10-20 by o6-nbeier