open62541/open62541 #2464: Problem to run discovery examples.

magiino, 2019-02-18:
## Description
I have problem with discovery_server_lds example. It can not write to socket.
[image]

## Background Information / Reproduction Steps
I have downloaded open62541 0.3.0 repo from https://open62541.org/ which is last release.
I generated VS 15 2017 project with this options.
[image]

In Visual Studio i run successfully build solution.
Then i run discovery_server_lds.exe -> see image up.

I have tried to run application as administrator, but it has the same result.

I have found similiar issue, which was already solved, but it could not read from socket - https://github.com/open62541/open62541/issues/1360

# Few more questions

1. Does single file distribution (open62541.c open62541.h) contains every feature defined in 0.3.0 release, or it contains just features which you select with CMAKE at generating project sources?
I did not select UA_ENABLE_AMALGAMATION, and i found, after build solution in VS, open62541.c/.h in root of build folder. Are they correct?

2. I'm new in creating bigger projects in C, especially with CMAKE. I want to use CLion IDE with mingw.
And i wonder, what is the best way to add open62541 to project. 

For now, I have just add single file distribution files to CMAKE like this
[image]

But I saw different approach here: https://github.com/Pro/open62541-arduino  and  here: https://stackoverflow.com/questions/54193481/build-open62541-library-as-external-library-with-cmake

Can you explain difference between my approach and those others?

THANKS!

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 0.3.0 (6249bb2)
 - [ ] Other OPC UA SDKs used (client or server): NONE
 - [x] Operating system: Windows 10.0.17134.590
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached: examples\discovery\discovery_server_lds.c
 - [ ] Critical issue

ccvca, 2019-02-19:
>     2\. I'm new in creating bigger projects in C, especially with CMAKE. I want to use CLion IDE with mingw.
>         And i wonder, what is the best way to add open62541 to project.

1. I would suggest using the latest Version of the master branch, as there has been some fixes to the install, ...

I'm currently using open62541 in CLion with mingw**-w64** (The normal MinGw does not work for me) and include it with find_package (https://cmake.org/cmake/help/v3.8/command/find_package.html) to my projects.

wenxinleong, 2019-02-19:
I'm running open62541 on raspberry Pi3 and I get error `Multicast error: Can not write to socket. Resource temporarily unavailable` , I initially though there is an error starting multicast DNS, but strangely if I continue with the example it works ok... maybe you can try that too. Launch the example in following order:
discovery_server_lds -> discovery_server_multicast -> discovery_client_find_servers

magiino, 2019-02-19:
Thanks guys for advice, but i moved today from win to ubuntu and everything is working like a charm.

closed 2019-02-19 by magiino