open62541/open62541 #2467: CMake CMP0054 warning when configuring for Visual Studio 2017

traversaro, 2019-02-19:
## Description
When configuring open62541 with a recent CMake on Windows with a Visual Studio generator, a warning related to [CMake's CMP0054 policy](https://cmake.org/cmake/help/v3.14/policy/CMP0054.html) is printed.

## Background Information / Reproduction Steps
This is the warning printed. tested using both CMake 3.10.3 and CMake 3.14-rc2 : 
~~~
CMake Warning (dev) at CMakeLists.txt:852 (if):
  Policy CMP0054 is not set: Only interpret if() arguments as variables or
  keywords when unquoted.  Run "cmake --help-policy CMP0054" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.

  Quoted variables like "MSVC" will no longer be dereferenced when the policy
  is set to NEW.  Since the policy is not set the OLD behavior will be used.
This warning is for project developers.  Use -Wno-dev to suppress it.
~~~

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): https://github.com/open62541/open62541/commit/18427523b4686685c5a94662c7107a68a1d09ec3
 - [x] Other OPC UA SDKs used (client or server): Not applicable
 - [x] Operating system: Windows
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached: Not applicable
 - [x] Wireshark network dump attached: Not applicable
 - [x] Self-contained code example attached: Not applicable
 - [x] Critical issue: No

Pro, 2019-02-25:
Can you create a PR which sets the policy to NEW? Thanks!

traversaro, 2019-02-28:
> Can you create a PR which sets the policy to NEW? Thanks!

Ok, I will prepare it in the next days.

marwinglaser, 2025-03-18:
Fixed by requiring newer minimum cmake version

closed 2025-03-18 by marwinglaser