open62541/open62541 #2468: Binary encoding for types derived from builtin types broken

jojastahl, 2019-02-19:
When using UtcTime datatype which is derived from DateTime, the encoding is broken. The `typeKind` field is set up with `UA_DATATYPEKIND_DATETIME`, then in line 998 of ua_types_encoding_binary.c https://github.com/open62541/open62541/blob/18427523b4686685c5a94662c7107a68a1d09ec3/src/ua_types_encoding_binary.c#L994-L998 the encoding is based on typeIndex, which is 47 for UtcTime. This results in encoding 48 (0x30), but should be 13.

I think in the code line above, `typeKind` should be used instead of `typeIndex`.

jpfr, 2019-02-19:
Is UtcTime printed as an immediate DateTime or wrapped in an ExtensionObject?
It is not a builtin type per-se...

jojastahl, 2019-02-19:
Is is not wrapped in an ExtensionObject.

OPC Part 3, 5.8.2 defines types that derive from built-in types as Simple Types:

> Simple DataTypes are subtypes of the Built-in DataTypes. They are handled on the wire like
the Built-in DataType, i.e. they cannot be distinguished on the wire from their Built-in
supertypes. Since they are handled like Built-in DataTypes regarding the encoding they cannot
have encodings defined in the AddressSpace. Clients can read the DataType Attribute of a
Variable or VariableType to identify the Simple DataType of the Value Attribute. An example of
a Simple DataType is Duration. It is handled on the wire as a Double but the Client can read
the DataType Attribute and thus interpret the value as defined by Duration (see 8.13).

commit 481c00e3 (2019-02-19): Stack: Use DataTypeKind everywhere instead of the index (fix #2468)
closed 2019-02-19 by jpfr