open62541/open62541 #2470: Heap Corruption when using custom datatypes from example

GLmontanari, 2019-02-19:
## Description
Dear all,
I tried to add a customized datatype to my server by slightly modifying the example code that I found here:  https://github.com/open62541/open62541/blob/master/examples/custom_datatype/server_types_custom.c

The code compiles, but I get a run-time error. Inspection of the code step by step with Visual Studio revealed that the program exits at this point:
```
void UA_initialize_architecture_network(void){
  WSADATA wsaData;
  WSAStartup(MAKEWORD(2, 2), &wsaData);
}
```

in the file `WinSock2.h` .

The call stack is as follows:
 	ntdll.dll!776f84ec()	Unknown
 	[Frames below may be incorrect and/or missing, no symbols loaded for ntdll.dll]	Unknown
 	[External Code]	
>	open62541_exercises.exe!UA_initialize_architecture_network() Line 199342	C
 	open62541_exercises.exe!ServerNetworkLayerTCP_start(UA_ServerNetworkLayer * nl, const UA_String * customHostname) Line 199673	C
 	open62541_exercises.exe!UA_Server_run_startup(UA_Server * server) Line 30862	C
 	open62541_exercises.exe!UA_Server_run(UA_Server * server, volatile bool * running) Line 30972	C
 	open62541_exercises.exe!main() Line 990	C++
 	[External Code]	

Sometimes instead, I can run the server but as soon as I connect to it with my client application (UAExpert) I get
[image]
While UAExpert reports "Read attributes of node 'NS1|String|3D.Point' failed [ret = Bad EncodingLimitsExceeded]"

Do you have any idea of how I could solve this?

## Background Information / Reproduction Steps
The code is the same as in the example, I have just copy-pasted it. The only difference is in the includes:
I have compiled the library using amalgamation, therefore substitute all the `#include` with `#include "open62541.h"`
## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 0.4.0
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Windows 10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-02-19:
Hmm. Can you tell us about the compiler and build flags?
Debug or Release build?

GLmontanari, 2019-02-20:
**Debug Win32**. Windows SDK 10.0.17134.0 Platform Toolset Visual Studio 2017 (v141)
**_MSC_VER 1912**

All options
`/JMC /permissive- /GS /analyze- /W3 /Zc:wchar_t 
/I"C:\Users\foo\Desktop\open62541\include" 

/I"C:\Users\foo\Desktop\open62541\build\src_generated" 

/I"C:\Users\foo\Desktop\open62541\arch\wec7" 

/I"C:\Users\foo\Desktop\open62541\plugins" 

/ZI /Gm- /Od /sdl /Fd"Debug\vc141.pdb" 
/Zc:inline /fp:precise /D "_CRT_SECURE_NO_WARNINGS" /D "_MBCS" 
/errorReport:prompt /WX- /Zc:forScope /RTC1 /Gd /Oy- /MDd /FC /Fa"Debug\" 
/EHsc /nologo /Fo"Debug\" /Fp"Debug\open62541_exercises.pch" /diagnostics:classic `

GLmontanari, 2019-02-20:
I'm sorry, I rebuilt the library using CMake and cloning from the master branch. This time I unchecked UA_ENABLE_AMALGAMATION and checked UA_BUILD_EXAMPLES.
Then I ran the example with the custom datatypes and it gave no problems...

jpfr, 2019-02-20:
Ok, great. Then I assume there was a version mismatch between the single-file library and the examples.
Please reopen if that is not the case.

closed 2019-02-20 by jpfr