open62541/open62541 #2474: Event propagation doesn't comply to spec

jojastahl, 2019-02-20:
As far as I understand the current codebase, when an event is triggered, it is added to all "HasComponent" parents of the source node. This doesn't comply with the specification, IMHO. The specs define a HasNotifier / HasEventSource hierarchy for event propagation, if my understanding of OPC UA Part 3, 7.17 and 7.18 is correct. In addition, 7.17 says that the Server singleton object should report all events generated anywhere, regardsless of any HasEventSource references.

Is my point of view correct, or am I missing something?

jpfr, 2019-02-20:
@Pro @aDogCalledSpot 
Can you confirm?

Can we make a meta-issue that collects things that need to be done for Events to get rid of the "experimental" status?

ccvca, 2019-02-20:
The current implementation is the behaviour, which I would expect after diskussion this with several people about this topic.

[image]
Speaks about that an Object ("Area") would deliver all Events from a subset, but it also says that this is just an example.

From 5.4:
[image]
So a mechanism to propagate events to parent nodes.

jojastahl, 2019-02-21:
I agree that OPC UA Part 3 is a little soft on how to handle event propagation. However I feel that the existence of HasNotifier/HasEventSource and the samples given for these references in Part 3, aswell as Part 9 (Alarms and Conditions) points heavily in the direction to respect these. That's my personal feeling.

I looked at how others are doing: I compared with the Softing OPC UA Toolkit using this model:

[image]

Result:

1. Raising an event on Alarms:
   - Softing: The event appears on Alarms and Server Object, 1x each.
   - Open62541: The event appears on Alarms, OPC-Test-System, DeviceSet and Server Object, 1x each
2. Raising an event on Conditions:
   - Softing: The event appears on Conditions, OPC-Test-System and Server Object, 1x each
   - Open62541: The event appears 1x on Conditions, and 2x on each of OPC-Test-System, DeviceSet and Server Object

Conclusion:
- Softing reports each event on the event source, and always on the server, too. Furthermore they are reported on objects in the HasNotifier/HasEventSource hierarchy.
- Open62541 reports on any parent which is enabled for events, and on the server. If there is a HasNotifier/HasEventSource reference, it is even reported twice.
- Open62541 reports an event raised on Conditions 2x on DeviceSet, even if there is no HasNotifier/HasEventSource link.

jpfr, 2020-06-23:
Was this solved with this commit last year?
https://github.com/open62541/open62541/commit/e6baba149f4f66d3ddb40baacb91354e2d812509#diff-dd7eb0316f0b90c03ec8a6123b429054

teccheck, 2025-04-05:
As there's no further reply, I'll close this issue. Feel free to reopen if needed

closed 2025-04-05 by teccheck