open62541/open62541 #2477: [OPCUA CLIENT]: hard to distinguish the same type of event fields in event subscription callback

elitezhu, 2019-02-21:
## Description
I use OPEN62541 as a OPCUA client and subscribe a event. when client receive a server response and jump into the event subscription callback. According to the example in
https://github.com/open62541/open62541/blob/master/examples/tutorial_client_events.c, it differs each filtered event field by event field type. But if in the fields, there are more than two ones with the same type, How can I parse each of them in the callback?

It's better to add a char* variable into the struct UA_Variant to indicate each event field name, I think.

## Background Information / Reproduction Steps

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): main branc(need encryption function)
 - [ ] Other OPC UA SDKs used (client or server): RF650R(server)
 - [ ] Operating system: windows
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-02-21:
You can disambiguate by the order of the fields.

Pro, 2019-02-25:
Another option would be to create two different monitored items/subscriptions and then use the monitoredItemId to distinguish the events.

The event handler callback is intentionally kept very simple, to allow the user to add more intelligence if necessary.

closed 2019-02-25 by Pro

elitezhu, 2019-03-01:
OK，thanks for your answer!
And another one, do we hava a API to transfer the error code to error string. So we can hunt the root cause more quickly.

elitezhu, 2019-03-04:
OK, thanks.