open62541/open62541 #2484: Incorrect usage of printf() formatting codes for UA_UInt32 and friends

pfalcon, 2019-02-21:
## Description

open62541 defines UA_UInt32 as:

~~~
typedef uint32_t UA_UInt32;
~~~

But then in numerous logging calls it prints values of the UA_UInt32 type using e.g. "%i". This is not correct and implicitly assumes that uint32_t is itself defined as "unsigned int". This doesn't have to be the case. And actually, there're various embedded toolchains which define it as "unsigned long". Then, building open62541 with such toolchains leads to dozens of warnings (which open62541's build system by default turns into errors).

Instead, the C standard defines special macros to deal with formatting of stdint.h types. See http://pubs.opengroup.org/onlinepubs/009696799/basedefs/inttypes.h.html for reference, and e.g. https://stackoverflow.com/questions/45922817/what-is-unquoted-priu32-in-printf-in-c as an informal intro.

In other words, code like:
~~~
processSequenceNumberAsym(UA_SecureChannel *channel, UA_UInt32 sequenceNumber) {
    UA_LOG_TRACE_CHANNEL(channel->securityPolicy->logger, channel,
                         "Sequence Number processed: %i", sequenceNumber);
~~~

should instead be:
~~~
processSequenceNumberAsym(UA_SecureChannel *channel, UA_UInt32 sequenceNumber) {
    UA_LOG_TRACE_CHANNEL(channel->securityPolicy->logger, channel,
                         "Sequence Number processed: " PRIu32, sequenceNumber);
~~~

## Background Information / Reproduction Steps

See above.

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): f90a74c2bec6c5ee8b3531d2d5b7895af1b30cd9
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Zephyr RTOS
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

pfalcon, 2019-02-21:
> should instead be:

Oh, given that open62541 doesn't directly use uint32_t, but provides indirection via UA_UInt32, it apparently should do the same for formatting codes too. So, should define something like `UA_PRIu32`.

jpfr, 2019-02-21:
We do directly use uint32_t.
It's just that we have a typedef for consistent naming.

UA_PRIu32 sounds good. Visual Studio is a bit tricky here if I recall correctly.

Do you want to make a PR that adds such a definition to /include/plugin_log.h?

pfalcon, 2019-02-22:
Thanks for response!

> Do you want to make a PR that adds such a definition to /include/plugin_log.h?

Well, I just was making a proof of concept port of open62541 to Zephyr RTOS https://github.com/zephyrproject-rtos/zephyr . It was mostly exploratory task to see where Zephyr falls short with POSIX compatibility, etc. (we're fixing those issues). As a good open-source citizen, I also report back portability issues I found in open62541, but I'm afraid, I won't be able to tackle them myself.

jpfr, 2019-02-22:
Fair enough. I put this on the milestone for 0.4.
And marked it as a good opportunity for new contributors.

DevTigerart, 2025-04-08:
Hi,
Is this issue still available ?
I would be interested in working on it as a first contribution on open62541.
Best regards