open62541/open62541 #2485: UA_UInt32, etc. isn't used consistenty, redefining isn't really possible

pfalcon, 2019-02-21:
## Description

This bugreport starts where https://github.com/open62541/open62541/issues/2484 left. Given that there're dozens of warnings when building open62541 with some toolchains, because open62541 implicitly assumes that `UA_UInt32` is defined as `unsigned int`, it's tempting to actually go and define it like that:

~~~
-typedef uint32_t UA_UInt32;
+typedef unsigned UA_UInt32;
~~~

However, this still leads to type mismatch warnings, because turns out, `UA_UInt32` isn't used consistently by open62541 itself. For example, UA_atomic_addUInt32 is defined as:

~~
static UA_INLINE uint32_t
UA_atomic_addUInt32(volatile uint32_t *addr, uint32_t increase) {
~~~

But as first argument, address of UA_UInt32 vars is actually passed.

This still leads to warnings like below:

~~~
/home/pfalcon/projects-3rdparty/Embedded/Zephyr/open62541demos/open62541ping/open62541/src/server/ua_session_manager.c: In function ‘UA_SessionManager_createSession’:
/home/pfalcon/projects-3rdparty/Embedded/Zephyr/open62541demos/open62541ping/open62541/src/server/ua_session_manager.c:159:25: warning: passing argument 1 of ‘UA_atomic_addUInt32’ from incompatible pointer type [-Wincompatible-pointer-types]
     UA_atomic_addUInt32(&sm->currentSessionCount, 1);
                         ^
In file included from /home/pfalcon/projects-3rdparty/Embedded/Zephyr/open62541demos/open62541ping/open62541/arch/ua_architecture_functions.h:205:0,
                 from /home/pfalcon/projects-3rdparty/Embedded/Zephyr/open62541demos/open62541ping/open62541/arch/posix/ua_architecture.h:140,
                 from /home/pfalcon/projects-3rdparty/Embedded/Zephyr/open62541demos/open62541ping/open62541/BUILD/src_generated/ua_config.h:101,
                 from /home/pfalcon/projects-3rdparty/Embedded/Zephyr/open62541demos/open62541ping/open62541/include/ua_types.h:19,
                 from /home/pfalcon/projects-3rdparty/Embedded/Zephyr/open62541demos/open62541ping/open62541/include/ua_server.h:16,
                 from /home/pfalcon/projects-3rdparty/Embedded/Zephyr/open62541demos/open62541ping/open62541/src/server/ua_session_manager.h:15,
                 from /home/pfalcon/projects-3rdparty/Embedded/Zephyr/open62541demos/open62541ping/open62541/src/server/ua_session_manager.c:12:
/home/pfalcon/projects-3rdparty/Embedded/Zephyr/open62541demos/open62541ping/open62541/arch/ua_architecture_definitions.h:319:1: note: expected ‘volatile uint32_t * {aka volatile long unsigned int *}’ but argument is of type ‘UA_UInt32 * {aka unsigned int *}’
 UA_atomic_addUInt32(volatile uint32_t *addr, uint32_t increase) {
 ^~~~~~~~~~~~~~~~~~~
~~~

## Background Information / Reproduction Steps

See above.

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): f90a74c2bec6c5ee8b3531d2d5b7895af1b30cd9
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Zephyr RTOS https://github.com/zephyrproject-rtos/zephyr
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

pfalcon, 2019-02-21:
@apalos, @muvarov: If/when you play with open62541 port, https://github.com/pfalcon/open62541, you'll see these warnings, and this ticket explains why.

jpfr, 2019-02-21:
On a 16bit platform, e.g. some microcontrollers, int is not 32bit long.
UA_UInt32 is always encoded with 4 byte on the network. So no, we cannot make this change.
Otherwise we risk data loss.

You mention that some embedded problems cause problems.
Can we cross-compile for them in our CI infrastructure?
Or build a linux executable against their header definitions....

Otherwise we cannot avoid breakage in the future.

pfalcon, 2019-02-22:
> So no, we cannot make this change.

Sorry if I didn't make it clear, but you don't need to make this change. This report is about different matter: open62541 seems to go a long way to allow to cover portability and flexibility needs. But when it actually comes to exercising those matters, e.g. an end user patches open62541 to redefine UA_UInt32 to something else, based on the peculiar needs of their platform, it all falls apart, because those "portability and flexibility needs" still not implemented consistently.

If there's something to do about this ticket, there would be one of two things:

* Make sure that open62541 use UA_UInt32, etc. 100% consistently, and never uses uint32_t (as mentioned, currently some functions, e.g. UA_atomic_addUInt32() declared as taking uint32_t, but are being passed UA_UInt32).
* Vice versa, remove UA_UInt32, etc., and instead use ANSI C standard types `uint32_t`, etc. directly (it's available in the standard C for 20 years now).

Note that I tried to change UA_atomic_addUInt32() to accept UA_UInt32, but turns out, that in its header, UA_UInt32 is not yet available.

muvarov, 2019-02-22:
@pfalcon I do not see that error. But my setup is a little bit hacky it's  linux posix platfrom with lwip network stack. From reading this issue I have no idea why open62541 invents new type  UA_UInt32. I think that any compiller supports uint32_t. If open62541 needs attomic variables then it makes sense to define them with corresponding access functions.

pfalcon, 2019-02-22:
@muvarov, Of course you don't. As was said, it happens only when someone tries to workaround another issue, #2484, with peculiar toolchains. Like, Zephyr toolchains, official ARM Embedded toolchain, etc.

E.g. ARM toolchain:
~~~
gcc-arm-none-eabi-7-2017-q4-major/bin$ ./arm-none-eabi-gcc -dM -E - </dev/null | grep UINT32_TYPE
#define __UINT32_TYPE__ long unsigned int
~~~

And again, the real bug is #2484. This one is report along the lines of "open62541 seems to provide flexibility with types, but in real usecases it falls short".

jpfr, 2019-02-22:
I don't follow the logic of open62541 not being consistent.
We use uint32_t every we want an unsigned 32bit integer. How is that not consistent?
And on 95% of platforms this is the standard way since at least C99.
https://en.cppreference.com/w/c/types/integer

If your platform does not provide uint32_t, then define it.
Instead of redefining UA_UInt32.

Same effort, consistent result.

jpfr, 2019-02-22:
Oh wow. The name just rang a bell.

@pfalcon Big fan of your work on micropython. 😄

pfalcon, 2019-02-26:
@jpfr: Thanks ;-).

> I don't follow the logic of open62541 not being consistent.
We use uint32_t everywhere we want an unsigned 32bit integer. How is that not consistent?

You don't. Majority of functions are declared as taking UA_UInt32, etc., and use corresponding variables, but some functions are declared as taking uint32_t, etc.

> If your platform does not provide uint32_t, then define it.
Instead of redefining UA_UInt32.

It provides, it just doesn't work well due to #2484. So, seeing UA_UInt32, I held onto possibility to override it to workaround #2484, but it didn't work fully either. And seems you don't even consider possibility to override UA_UInt32 a valid usecase.

That's fine. All in all, this is pretty peculiar and borderline issue, I wouldn't venture into that if #2484 worked. It's also a number game - one report means nothing, but if more people report similar issues, maybe something can be improved. So I hope, it will be possible to find this issue later.

Otherwise, closing to avoid further confusion. Thanks for the discussion!

closed 2019-02-26 by pfalcon