open62541/open62541 #2491: high-level functions does not have function to obtain timestamp along with the value

JSkrat, 2019-02-22:
## Description
One can simply read node value using UA_Client_readValueAttribute function. But if one want to read server and source timestamps along with the value, need to copy UA_Client_readValueAttribute function with slight modifications. 

`request.timestampsToReturn = UA_TIMESTAMPSTORETURN_SERVER;`

Proposition is to add high-level functions to obtain those timestamps along with the data, atomically.

## Background Information / Reproduction Steps

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 0.3.0 (amalgam)
 - [ ] Other OPC UA SDKs used (client or server): client
 - [ ] Operating system: ubuntu
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-02-22:
This sounds like a very shallow wrapper around `UA_Client_Service_read`.
https://open62541.org/doc/current/client.html#services

Why don't you try your hand in implementing such a method?

JSkrat, 2019-04-02:
what branch should i use for making changes? shall it be pullrequest?

JSkrat, 2019-04-02:
and is there some function naming conventions i should know?

ccvca, 2019-04-02:
When you want to submit code to this project you need to do a Pull-Request. (Fork and then work on a separate branch in your project).

For the function name comvention, have a look at the existing functions.
Public functions follow the scheme: UA_<Object/Group>_<camelCaseName> (e.g. UA_Timer_deleteMembers)

JSkrat, 2019-04-02:
thanks for the reply. but i have more questions)
1. can i add new argument to a __UA_Client_readAttribute and update all functions using it, or should i create a macro so the one can call that function in old way? or should i create new one, like __UA_Client_readAttributeWithTimestamps and copypaste the whole set of using __UA_Client_readAttribute functions (sounds ugly to me, but who knows, maybe there is some compatibility i don't see)
2. can i modify the signature of highlevel functions so they will take one more argument for the timestamps? or copypaste all of them with suffixed names? what is the correct approach here? i don't want to break any existing code
3. shall i modify/add new set of all highlevel functions at once, or one i actually need now (UA_Client_readValueAttribute) would be sufficient?

sorry, i'm not very experienced in contributing

ccvca, 2019-04-03:
I would always prefer non breaking changes, but as I'm not a maintainer maybe @jpfr could add a comment too.
My opinion would be:

1. As __UA_Client_readAttribute is an internal function, adding an additional argument seems fine too me.
2. A change in the UA_Client_readAttribute-function(s) would brakt many implementations, so I would preffer a "extended Version" of it. Instead of an additional attribute I would suggest using the UA_DataValue (see code below) instead of UA_Variant.
3. As timestamps could be returned for any attribute, it is consequent to provide it for all attributes.

https://github.com/open62541/open62541/blob/1b14ad1d28551cf4cc613199d20101fbfb102bd3/include/open62541/types.h#L712-L731

Pro, 2019-04-03:
> I would always prefer non breaking changes, but as I'm not a maintainer maybe @jpfr could add a comment too.

Correct! Especially breaking changes to the public API  should be avoided if possible.

> My opinion would be:
> 
>     1. As __UA_Client_readAttribute is an internal function, adding an additional argument seems fine too me.

Maybe you can create a new method called something like `__UA_Client_readAttributeDataValue`?

> 
>     2. A change in the UA_Client_readAttribute-function(s) would brakt many implementations, so I would preffer a "extended Version" of it. Instead of an additional attribute I would suggest using the UA_DataValue (see code below) instead of UA_Variant.

Agree. First try to propose a Pull Request which only adds a `UA_Client_readValueAttribute_time` (maybe there is a better naming for it).
Based on that we can think if it makes sense to support timestamps for other attributes.

> 
>     3. As timestamps could be returned for any attribute, it is consequent to provide it for all attributes.

JSkrat, 2019-04-09:
okay. now i'm struggling with built library. I used amalgam Git-Revision: v0.3.0 in my project before. I decided to check new changes on it, so i changed headers to a new library and eventually it compiled successfully. But it just refuses to connect to my running opcua server on the same library (amalgam too). It compiles and kinda creates a client, but returns 80020000 at connection

I wonder if any of you aware of some obvious changes i should've made to my project in order to "migrate" from 0.3.0 to the recent one from the master (commit 217eed40cf753a246ebb52917aa1d77800ac5f7c (HEAD -> add_node_timestamp_to_highlevel_functions_rezult, origin/master, origin/HEAD, master)
Author: Jayanth Velusamy <jayanth.v@kalycito.com>
Date:   Fri Mar 29 12:20:39 2019 +0530)

I want to check my changes 100% not crashing anything and works before pulling 9_9

JSkrat, 2019-04-10:
and i not really understand how do i turn logs there... in official amalgam i could just change UA_LOGLEVEL in header to 100 to see everything and it worked. But here, in generated amalgam, i'm changing it in ccmake dialog (tried to turn UA_DEBUG on just in case too), re-generating, it really generates config with UA_LOGLEVEL 100, but there are no any messages in the console. Am i doing something wrong?

in code UA_Client_new() (UA_ClientConfig_default is not working anymore for some reason) all config just filled up with zeroes by memset and never filled up.

maybe there is some way to explicitly turn on some plugins?

sorry for flood, but do not know any better place to communicate(

jpfr, 2019-06-06:
Can you just try a fresh checkout?
The docs should guide you to a working example.

Please reopen if the activity picks up again.

closed 2019-06-06 by jpfr