open62541/open62541 #2492: Building EUROMAP 83 generated nodeset produces multiple 'incompatible types' errors

GeorgeSellner, 2019-02-22:
## Description
**Files from euromap.org for EUROMAP 83, after generation to .c/.h generates 'incompatible types' errors**

## Background Information / Reproduction Steps
Download files from [http://www.euromap.org/en/euromap83/](http://www.euromap.org/en/euromap83/)
Use UAModeler to generate .
[gen_euromap77a.bash.txt](https://github.com/open62541/open62541/files/2895846/gen_euromap77a.bash.txt)
[CMakeLists.txt](https://github.com/open62541/open62541/files/2895854/CMakeLists.txt)
[open62541_generated.zip](https://github.com/open62541/open62541/files/2895861/open62541_generated.zip)

bsd files
generate nodeset code using nodeset_compiler.py, generate_datatypes.py
Upon make, get following output:
`$ make
[ 11%] Building C object CMakeFiles/opcua_server.dir/open62541_generated/ua_namespace_euromap83.c.o
/home/mmtools/workspace/euromap77a_open62541/open62541_generated/ua_namespace_euromap83.c: In function ‘function_ua_namespace_euromap83_276_begin’:
/home/mmtools/workspace/euromap77a_open62541/open62541_generated/ua_namespace_euromap83.c:9907:78: error: incompatible types when assigning to type ‘size_t {aka unsigned int}’ from type ‘UA_String {aka struct <anonymous>}’
 ode_ns_2_i_6104_ProductionDatasetInformationType_0->productNameSize = UA_STRING("");
                                                                     ^
/home/mmtools/workspace/euromap77a_open62541/open62541_generated/ua_namespace_euromap83.c: In function ‘function_ua_namespace_euromap83_378_begin’:
/home/mmtools/workspace/euromap77a_open62541/open62541_generated/ua_namespace_euromap83.c:13720:78: error: incompatible types when assigning to type ‘size_t {aka unsigned int}’ from type ‘UA_String {aka struct <anonymous>}’
 ode_ns_2_i_6136_ProductionDatasetInformationType_0->productNameSize = UA_STRING("");
                                                                     ^
/home/mmtools/workspace/euromap77a_open62541/open62541_generated/ua_namespace_euromap83.c: In function ‘function_ua_namespace_euromap83_502_begin’:
/home/mmtools/workspace/euromap77a_open62541/open62541_generated/ua_namespace_euromap83.c:18462:78: error: incompatible types when assigning to type ‘size_t {aka unsigned int}’ from type ‘UA_String {aka struct <anonymous>}’
 ode_ns_2_i_6276_ProductionDatasetInformationType_0->productNameSize = UA_STRING("");
                                                                     ^
CMakeFiles/opcua_server.dir/build.make:114: recipe for target 'CMakeFiles/opcua_server.dir/open62541_generated/ua_namespace_euromap83.c.o' failed

make[2]: *** [CMakeFiles/opcua_server.dir/open62541_generated/ua_namespace_euromap83.c.o] Error 1
CMakeFiles/Makefile2:72: recipe for target 'CMakeFiles/opcua_server.dir/all' failed
make[1]: *** [CMakeFiles/opcua_server.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2`
## Checklist

Branch: master
Commit Id: 45ca322125c3eb669705070d177f65d1d4f8cebe
Linux MMLCA1-5509 4.15.0-45-generic #48-Ubuntu SMP Tue Jan 29 16:28:13 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

Please provide the following information:

 - [x] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [x] Critical issue

Pro, 2019-02-25:
Probably the value which you are using for `--namespace` passed to `generate_datatypes.py` is not correct, therefore the type node ids are not correctly mapped.
I.e. you are mapping DI to NS 1, but it is probably on NS 2 on the running server?
E.g. see here: https://github.com/open62541/open62541/blob/e2df239750f2f8cbcbd762c0540e6a95e6b82724/examples/nodeset/CMakeLists.txt#L56

Try to shift all the values by +1. Does that work?

This is still an open issue and the datatypes generation should not require the namespace index.

GeorgeSellner, 2019-02-25:
Thank you @Pro 

I made the suggested change to calls to `generate_datatypes.py`, shifting all the values by +1
Unfortunately I still receive the same errors as before `error: incompatible types`

Pro, 2019-02-25:
Ah, I just read your issue description again and saw that my previous comment was not relevant to the error (but still you should use the +1 shifted values).

I managed to get the euromap 83 nodeset up and running.

First of all you should thank @matkonnerth who just created the PR #2366 which exactly fixes your issue above.

Additionally the .csv file provided by euromap is missing some node ids which I had to add manually:
https://github.com/Pro/open62541/commit/8ccd84ccbb1bb326eca59ea08a611e543b8709e0

You can check out my branch here, which has an example for the euromap server:
https://github.com/Pro/open62541/tree/test/euromap

Especially the folder:
https://github.com/Pro/open62541/tree/test/euromap/examples/nodeset/euromap

Please reopen if you still have issues compiling the nodesets.

closed 2019-02-25 by Pro

Pro, 2019-02-26:
@GeorgeSellner #2504 will add an additional error output if the .csv is missing the corresponding node ids. Fixes https://github.com/Pro/open62541/commit/8ccd84ccbb1bb326eca59ea08a611e543b8709e0

GeorgeSellner, 2019-02-26:
Hi Stefan @Pro 

Not sure if I have the right commit point, but when I try ccmake to enable building the examples get following error:

<pre> CMake Error at examples/CMakeLists.txt:29 (add_executable):
   Cannot find source file:

     /home/mmtools/ws_open/open62541/build/src_generated/ua_namespace_di.c

   Tried extensions .c .C .c++ .cc .cpp .cxx .cu .m .M .mm .h .hh .h++ .hm
   .hpp .hxx .in .txx
 Call Stack (most recent call first):
   examples/nodeset/euromap/CMakeLists.txt:28 (add_example)

 CMake Error at examples/CMakeLists.txt:29 (add_executable):
   No SOURCES given to target: server_nodeset_euromap
 Call Stack (most recent call first):
   examples/nodeset/euromap/CMakeLists.txt:28 (add_example)
</pre>

Pro, 2019-02-26:
Fixed. Please update your local copy of the `test/euromap` branch. The generate macros somehow don't like it if they are in different cmake files.

Then you can run the example with:

```bash
cd open62541
mkdir build
cd build
cmake -DUA_NAMESPACE_ZERO=FULL -DUA_BUILD_EXAMPLES=ON ..
make -j
./bin/examples/server_nodeset_euromap
```

GeorgeSellner, 2019-02-26:
Stefan,

All is good

Thank you!

GeorgeSellner, 2019-12-05:
Hi @Pro ,

Unfortunately Euromap have revised their files again.
[https://www.euromap.org/en/euromap77](https://www.euromap.org/en/euromap77)
[https://www.euromap.org/en/euromap83/](https://www.euromap.org/en/euromap83/)

I tried updating the example, but I'm getting errors from tools/generate_datatypes.py

output or error after $ make -j
<pre>[  8%] <font color="#729FCF"><b>Generating ../../src_generated/open62541/types_euromap83_generated.c, ../../src_generated/open62541/types_euromap83_generated.h, ../../src_generated/open62541/types_euromap83_generated_handling.h, ../../src_generated/open62541/types_euromap83_generated_encoding_binary.h</b></font>
[  8%] Built target open62541-generator-ids-powerlink
[  8%] Built target open62541-generator-ids-di
[  9%] Built target open62541-generator-namespace
[  9%] Built target open62541-generator-types-powerlink
[  9%] Built target open62541-generator-ids-testnodeset
[  9%] Built target open62541-generator-types-di
Traceback (most recent call last):
  File &quot;/home/mm-prog/open62541/open62541/tools/generate_datatypes.py&quot;, line 537, in &lt;module&gt;
    parseTypeDefinitions(outname, f, args.namespace)
  File &quot;/home/mm-prog/open62541/open62541/tools/generate_datatypes.py&quot;, line 359, in parseTypeDefinitions
    raise RuntimeError(&quot;Infinite loop detected trying to processing types &quot; + name + &quot;: unknonwn subtype &quot; + str(unknownTypes(typeXml)))
RuntimeError: Infinite loop detected trying to processing types ConfigurationParameterType: unknonwn subtype [&apos;EUInformation&apos;]
examples/nodeset/CMakeFiles/open62541-generator-types-euromap83.dir/build.make:65: recipe for target &apos;src_generated/open62541/types_euromap83_generated.c&apos; failed
make[2]: *** [src_generated/open62541/types_euromap83_generated.c] Error 1
CMakeFiles/Makefile2:2281: recipe for target &apos;examples/nodeset/CMakeFiles/open62541-generator-types-euromap83.dir/all&apos; failed
make[1]: *** [examples/nodeset/CMakeFiles/open62541-generator-types-euromap83.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[  9%] Built target open62541-generator-types-testnodeset
[ 17%] Built target open62541-plugins
[ 19%] Built target  open62541-generator-ns-di
[ 46%] Built target open62541-object
Makefile:151: recipe for target &apos;all&apos; failed
make: *** [all] Error 2
</pre>

Here is a link to what I have been trying.
[https://github.com/GeorgeSellner/open62541/tree/example/euromap77](https://github.com/GeorgeSellner/open62541/tree/example/euromap77)

Odd that it is saying it can't find **EUInformation**, I thought 'FULL' in cmake would take care of that?

Pro, 2019-12-06:
The issue here is that EUInformation is from the NS0 Types. This means that you also need to import that types.bsd file while working with the Euromap bsd.

I improved the error message and added the possibility to do this via the generate_namespace_and_datatypes macro.
See: #3314

And here is an example that works for me:
https://github.com/Pro/open62541/tree/example/euromap77

GeorgeSellner, 2019-12-06:
Hi @Pro 
Thank you so much Stefan!

Works for me.

ivan-ushakov, 2022-04-28:
@Pro
Sorry for bumping this again but I'm facing problem with importing `euromap83-v1.0.3`:
```
[2022-04-28 16:52:04.491 (UTC+0200)] warn/server	Writing the value of Node ns=3;i=6373 failed with the following reason: DataType of the value is incompatible
[2022-04-28 16:52:04.492 (UTC+0200)] info/session	SecureChannel 0 | Session "Administrator" | WriteRequest returned status code BadTypeMismatch
[2022-04-28 16:52:04.492 (UTC+0200)] info/session	SecureChannel 0 | Session "Administrator" | AddNode (ns=3;i=6373): The value is incompatible with the variable definition
[2022-04-28 16:52:04.492 (UTC+0200)] info/session	SecureChannel 0 | Session "Administrator" | AddNode (ns=3;i=6373): Type-checking failed with error code BadTypeMismatch
```
I use `NAMESPACE_IDX 2` for the DI and `NAMESPACE_IDX 3` for the EuroMap.

ivan-ushakov, 2022-04-28:
@GoetzGoerisch 
I found that problem was with `NAMESPACE_MAP`. When I provided correct value (starting from 2) error gone.

GoetzGoerisch, 2022-04-29:
Great! 
May I ask, for whom you are implementing this? Drop me a line by mail (see profile), if you don't want to write it here.

shaohuazhang1, 2022-12-21:
where is NAMESPACE_MAP and how can i change the value？

GoetzGoerisch, 2023-01-09:
The `NAMESPACE_MAP` is defined in your `CMakeLists.txt`.

Reference Server for EM83:
https://github.com/open62541/open62541-nodeset-loader/blob/cefa5c32e71caf80e1c8ceb03ea6c44664ec5a52/backends/open62541/tests/integration/reference_server/CMakeLists.txt