open62541/open62541 #2501: Dynamically create variable nodes during runtime

usman332211, 2019-02-25:
Hi all, I am looking for Dynamically Node addition, I am running my server and sometimes i want to update my adressspace (add or delete nodes). Is this possible?

jpfr, 2019-02-25:
This is possible. We support the AddNodes service.
This is a standard service in the OPC UA standard.

But you might need a plugin for UAExpert.
I'm not sure it supports these services.

Or you wrap it in a custom method you add to the information model.

usman332211, 2019-02-26:
@jpfr  thank you , can you bit explain the use of AddNodes, of link to example, if it is possible.

juangburgos, 2019-02-27:
Just call the `UA_Server_addVariableNode` after the server is running. The example code for using `UA_Server_addVariableNode` is in [the tutorials](https://open62541.org/doc/0.3/tutorial_server_variable.html).

In UAExpert, after adding a node when the connection is already active, you have to click the little **refresh** button in the *Address Space* plane:

[image]

For deleting nodes, there appear to be a `UA_Server_deleteNode`, but I have not used it. Though it seems simple, just provide the NodeId to delete.

Pro, 2019-03-05:
Not sure if you can do that with UAExpert. It's better to ask this in the UAExpert support forum. Our server supports adding nodes, as @jpfr already mentioned.

Here's an example using our client implementation:
https://github.com/open62541/open62541/blob/master/examples/client.c#L192

closed 2019-03-05 by Pro