open62541/open62541 #2503: AddNodes: No TypeDefinition, WriteRequest returned status code BadNotWritable

lakshmisivareddy, 2019-02-26:
Hi i am trying to write a client and server sample 
i am trying to do reading data from server and send data to server 
client able to read data from server but when client trying to write data using 'UA_Client_writeValueAttribute' i am always getting the error 

client log:
[2019-02-26 14:07:03.095 (UTC+0000)] info/client	Selected UserTokenPolicy open62541-anonymous-policy with UserTokenType Anonymous and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
Created 'NewVariable' with numeric NodeID 30305
Error Val -2143617024
data recieved from server : 34
data recieved from server : 36
data recieved from server : 38
data recieved from server : 40

server log:
2019-02-26 14:07:03.095 (UTC+0000)] info/channel	Connection 5 | SecureChannel 1 | Opened SecureChannel
[2019-02-26 14:07:03.096 (UTC+0000)] info/session	Connection 5 | SecureChannel 1 | Session ns=1;g=74eedb6e-d6e9-eb16-3533-530568a62ac2 | ActivateSession: Session activated
[2019-02-26 14:07:03.096 (UTC+0000)] info/session	Connection 5 | SecureChannel 1 | Session ns=1;g=74eedb6e-d6e9-eb16-3533-530568a62ac2 | AddNodes: No TypeDefinition for ns=1;i=30305; Use the default TypeDefinition for the Variable/Object
[2019-02-26 14:07:03.096 (UTC+0000)] info/session	Connection 5 | SecureChannel 1 | Session ns=1;g=74eedb6e-d6e9-eb16-3533-530568a62ac2 | WriteRequest returned status code BadNotWritable
data recieved from client :  0
Counter updated: 35
Counter updated: 36
data recieved from client :  0
Counter updated: 37

i am attaching the code 

**client:**

#include <ua_client_highlevel.h>
#include <ua_config_default.h>
#include <ua_log_stdout.h>

void addServerNode(void* ptr)
{
   UA_Client *client=ptr;
   UA_NodeId nodeId = UA_NODEID_NUMERIC(1, 30305);
   UA_QualifiedName counterName = UA_QUALIFIEDNAME(1, "Counter");
   UA_VariableAttributes attr = UA_VariableAttributes_default;
   attr.description = UA_LOCALIZEDTEXT("en_US","Server  Counter val");
   attr.displayName = UA_LOCALIZEDTEXT("en_US","Server Counter");
   attr.dataType = UA_TYPES[UA_TYPES_INT32].typeId;
   UA_Int32 counterValue = 0;
   UA_Variant_setScalarCopy(&attr.value, &counterValue, &UA_TYPES[UA_TYPES_INT32]);
   UA_StatusCode retval=UA_Client_addVariableNode(client,nodeId,
                            UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
                            UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
                           counterName, UA_NODEID_NULL, attr, &nodeId);
   if(retval == UA_STATUSCODE_GOOD )
        printf("Created 'NewVariable' with numeric NodeID %u\n", nodeId.identifier.numeric);

}

int main(void) 
{
    UA_Client *client = UA_Client_new();
    UA_ClientConfig *cc = UA_Client_getConfig(client);
    UA_ClientConfig_setDefault(cc);
    //cc->stateCallback = stateCallback;
    //cc->subscriptionInactivityCallback = subscriptionInactivityCallback;

    UA_StatusCode retval = UA_Client_connect(client, "opc.tcp://192.168.182.135:4840");
    if(retval != UA_STATUSCODE_GOOD) {
        UA_Client_delete(client);
        return (int)retval;
    }
    addServerNode(client);
    UA_Variant value; 
    UA_Variant_init(&value);

    while(1)
    {

        UA_Int32 temp = 0;
        const UA_NodeId nodeId = UA_NODEID_NUMERIC(1, 20305);
        retval = UA_Client_readValueAttribute(client, nodeId, &value);
        if(retval == UA_STATUSCODE_GOOD) {
                temp=*(UA_Int32*)value.data;
                printf("data recieved from server : %d\r\n",temp);
        }
        UA_sleep_ms(1000);
       const UA_NodeId nodeId1 = UA_NODEID_NUMERIC(1, 30305);
       UA_Variant *myVariant = UA_Variant_new();
       UA_Variant_setScalarCopy(myVariant, &temp, &UA_TYPES[UA_TYPES_INT32]);
       retval= UA_Client_writeValueAttribute(client,nodeId1, myVariant);
       if(retval == UA_STATUSCODE_GOOD)
       {
          printf(" UA_STATUSCODE_GOOD \r\n");
       }
       else
       {
           printf("Error Val %d\r\n",retval);
       }
       UA_Variant_delete(myVariant);
    }

    /* Clean up */
    UA_Variant_clear(&value);
    UA_Client_delete(client); 
    return UA_STATUSCODE_GOOD;
}

**server**

#include <ua_server.h>
#include <ua_config_default.h>
#include <ua_log_stdout.h>
#include <unistd.h>
 #include <time.h>
#define SLEEP_TIME_MILLIS 500
#define COUNTER_NODE_ID 20305
UA_Boolean running = true;
int32_t numberOfParts = 0;

static void stopHandler(int sig) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "Received ctrl-c");
    running = false;
}
static void addCounterSensorVariable(UA_Server * server) {

        UA_NodeId counterNodeId = UA_NODEID_NUMERIC(1, 20305);
        UA_QualifiedName counterName = UA_QUALIFIEDNAME(1, "Counter");
        UA_VariableAttributes attr = UA_VariableAttributes_default;
        attr.description = UA_LOCALIZEDTEXT("en_US","Client Counter Val");
        attr.displayName = UA_LOCALIZEDTEXT("en_US","Client Counter");
		attr.dataType = UA_TYPES[UA_TYPES_INT32].typeId;
        UA_Int32 counterValue = 0;
       UA_Variant_setScalarCopy(&attr.value, &counterValue, &UA_TYPES[UA_TYPES_INT32]);
        UA_Server_addVariableNode(server, counterNodeId,
                UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
                UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
                counterName, UA_NODEID_NULL, attr, NULL, NULL);
}
void *mainSensor(void *ptr) {
        UA_Server * server = ptr;
        int utime = SLEEP_TIME_MILLIS * 1000;
        while (1){
                numberOfParts = numberOfParts + 1;
                printf("Counter updated: %i\n", numberOfParts);
                UA_Variant value;
                UA_Int32 myInteger = (UA_Int32) numberOfParts;
                UA_Variant_setScalarCopy(&value, &myInteger, &UA_TYPES[UA_TYPES_INT32]);
                UA_Server_writeValue(server, UA_NODEID_NUMERIC(1,COUNTER_NODE_ID), value);
                usleep(utime);
        }
}
void *readDadafromClient(void* ptr)
{
   UA_Server * server = ptr;
   UA_Variant value;
    UA_Variant_init(&value);

  while(1)
  {
     const UA_NodeId nodeId = UA_NODEID_NUMERIC(1, 30305);
      UA_StatusCode retval=UA_Server_readValue(server,nodeId,&value);
     if(retval == UA_STATUSCODE_GOOD)
     {
        UA_Int32 temp = 0;
        temp=*(UA_Int32*)value.data;
        printf("data recieved from client :  %d\r\n",temp);

     }
     UA_sleep_ms(1000);
  }
  UA_Variant_clear(&value);
}
int main(void)
{
    pthread_t threadSensor;
    pthread_t tempratureSensor;
    signal(SIGINT, stopHandler);
    signal(SIGTERM, stopHandler);
    UA_ServerConfig *config = UA_ServerConfig_new_default();
    UA_Server *server = UA_Server_new(config);
    addCounterSensorVariable(server);
    if(pthread_create( &threadSensor, NULL, mainSensor, server)) {
                exit(EXIT_FAILURE);
        }
     if(pthread_create(&tempratureSensor,NULL,readDadafromClient,server))
       {
                exit(EXIT_FAILURE);
       }
    UA_StatusCode retval = UA_Server_run(server, &running);
    UA_Server_delete(server);
    UA_ServerConfig_delete(config);
    return (int)retval;
}

SiggyBar, 2020-01-09:
I got the same error message after updating open62541 from version 0.2 to 1.0.  The error was fixed by enabling write access like this:
`attr.accessLevel = UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE;`

MichaelKlin, 2020-09-23:
@lakshmisivareddy  Do you resolved your problem? I get the same error

MichaelKlin, 2020-09-23:
@SiggyBar also with giving the acces rights it works

cguimaraes, 2022-04-06:
@SiggyBar solution worked for me.
@EvgKlin I believe you should do it on the server side.

teccheck, 2025-04-05:
It seems to me like there's a working solution. Other than that, the relevant API has changed a lot since then, so I'll close this issue

closed 2025-04-05 by teccheck