open62541/open62541 #2505: UA_NodeId_hash Fail!

juangburgos, 2019-02-27:
## Description

I believe hashes are supposed to be functions that give unique output values for the input values. I suppose that is the pupose of the `UA_NodeId_hash` function. Nevertheless it **fails** on a very simple clase (see *Reproduction Steps* below).

## Reproduction Steps

```c++
#include <assert.h>
#include "open62541.h"

int main(void) 
{
	UA_NodeId nodeId1;
	nodeId1.namespaceIndex     = 1;
	nodeId1.identifierType     = UA_NODEIDTYPE_NUMERIC;
	nodeId1.identifier.numeric = 319;

	UA_UInt32 hashNodeId1 = UA_NodeId_hash(&nodeId1);

	UA_NodeId nodeId2;
	nodeId2.namespaceIndex     = 1;
	nodeId2.identifierType     = UA_NODEIDTYPE_NUMERIC;
	nodeId2.identifier.numeric = 320; // DIFFERENT !

	UA_UInt32 hashNodeId2 = UA_NodeId_hash(&nodeId2);

	// FAILS !
	assert(hashNodeId1 != hashNodeId2);

	return;
}
```

## Checklist

Please provide the following information:

- [x] open62541 Version (release number or git tag): master
- [ ] Other OPC UA SDKs used (client or server): 
- [x] Operating system: ANY
- [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
- [ ] Wireshark network dump attached
- [x] Self-contained code example attached
- [x] Critical issue

jpfr, 2019-02-27:
It is not possible that the hash is always unique.

The hash has 32bit.
With a string or guid NodeId, there more than 32bit of information.
So there must be duplicates.

But yeah, this example is suspicious.
Will investigate.

basyskom-meerkoetter, 2019-03-02:
had a quick look into it.
Observation:
its reproduceable.
the hash value is "198" for both nodes.
it works as expected with 318/319 or 320/321.

closed 2021-09-23 by jpfr