open62541/open62541 #2509: Network Layer start-up result ignored (errors get unnoticed...)

pnikiel, 2019-02-27:
## Description
Result of Network Layer start is ignored (for example because of TCP/IP port being already in use).
This gives false impression that an OPC-UA server started.

## Background Information / Reproduction Steps
Just try with a simplest possible server:
1. run the first instance (should start just fine),
2. in another terminal, run the second instance
you will see that UA_Server_run or UA_Server_run_startup (depending which one you used) absolutely ignored that TCP/IP port is already taken.

I inspected the source code and the problem comes from function:
void addServerSocket(ServerNetworkLayerTCP *layer, struct addrinfo *ai)
in arch/ua_network_tcp.c
which should return integer instead of void.
This function is called by ServerNetworkLayerTCP_start which is already made to return proper error code but it of course ignores the result of invocation to aforementioned addServerSocket.

By looking at the code I expect the problem to be both in current master as well as in 0.3.0; in my development env where I run things I clearly see this in 0.3.0.

 If there's interest from open62541 management, I could improve it.

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 0.3.0  and current master
 - [ ] Other OPC UA SDKs used (client or server):  none
 - [ ] Operating system:  Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached  below
 - [ ] Wireshark network dump attached   Not necessary
 - [ ] Self-contained code example attached   Just use the minimal server example.
 - [ ] Critical issue

Logs:
[2019-02-27 22:06:58.723 (UTC+0100)] warn/network	Error binding a server socket: Address already in use
[2019-02-27 22:06:58.723 (UTC+0100)] warn/network	Error binding a server socket: Address already in use
[2019-02-27 22:06:58.723 (UTC+0100)] info/network	TCP network layer listening on opc.tcp://pcatldcsp11.cern.ch:4841/

I.e. the error was noticed, the server shouldn't have started but it did.

Pro, 2019-02-28:
Thanks for reporting this!

Can you propose a PR with your changes?

pnikiel, 2019-02-28:
I will. Let's talk soon. Best Regards!

Infinity95, 2019-03-03:
This should already be handled differently in #2408.

pnikiel, 2019-03-12:
Hi, 
could either of you advise me if I should wait for #2408 and retest/propose improvement then, or I go ahead and propose a PR now?

Pro, 2019-03-12:
@Infinity95 do you have an estimate when #2408 will approx be ready to be merged?

Infinity95, 2019-03-12:
@Pro I'm waiting for a review on the PR and need to fix an issue with OSX and Windows, and then it should be ready imo.
Do you have any experience with OSX? If so, maybe you could have a look at the CI build and help me out. I'm out of ideas of what could be the cause of the compiler error.
On Windows there seem to be a few memory issues which i will have to debug.

pnikiel, 2019-12-17:
Dear @Pro, @mlgiraud ,
I retested with "v1.0" and the problem is still there.
Do you think right now the situation is stable enough that I could propose the fix?
Or I better try to fix on another base version? Would you give me an advice?

Pro, 2019-12-17:
you can propose a PR on the master branch. I guess #2408 will take some more time.  Later on we can integrate/rebase #2408 onto your changes.

pnikiel, 2020-02-26:
Hi, so the fix is merged now and this issue can be successfully closed.
Thank you for your help open62541 gurus ;-)

pnikiel, 2020-02-26:
OK I did something wrong ;-) How do I close it with 'success' return code? ;-)

SiggyBar, 2020-05-12:
I am running into the same problem, but the merged fix alone does not make the "Error binding a server socket: Address already in use" error happen for me like I want (both for Windows 10 and Ubuntu 18.04).

In addition to the merged fix, I also need to remove the code that sets the `SO_REUSEADDR` socket option in `addServerSocket()` to get the wanted behavior.

pnikiel, 2020-05-13:
hmm, maybe SO_REUSEADDR is something new that was added after my fix? Could you tell us which commit of open62541 you used?

SiggyBar, 2020-05-14:
I am on the `1.0` branch (commit d5df1c62) with a97cdb3 cherry-picked.  Setting the `SO_REUSEADDR` socket option was done long before your fix (you modified the code that sets this option to return `UA_STATUSCODE_BADCOMMUNICATIONERROR` when it fails).

pnikiel, 2020-05-18:
Hi @SiggyBar ,
I just tested with 1.1-rc1 and it works as expected on my system (CentOs 7.7, basically same as RHEL):

[2020-05-18 12:32:22.859 (UTC+0200)] warn/network	Error binding a server socket: Address already in use
2020-05-18 12:32.22.859378 [BaseQuasarServer.cpp:156, ERR] Exception caught in BaseQuasarServer::serverRun:  [UA_Server_run_startup returned not-good, server can't start. Error was:BadCommunicationError (0x80050000)]

Would you be able to test on 1.1-rc1?
I'm very intrigued on where the problem might be.

SiggyBar, 2020-06-05:
I finally got to test this with 1.1 (commit bcec6028).  The results are the same as before.  Starting two OPC UA servers with the same port produces no error before I make this change to `addServerSocket()`:
```c
    if(UA_setsockopt(newsock, SOL_SOCKET, SO_REUSEADDR,
                  (const char *)&optval, sizeof(optval)) == -1) {
        UA_LOG_WARNING(layer->logger, UA_LOGCATEGORY_NETWORK,
                       "Could not make the socket reusable");
        UA_close(newsock);
        return UA_STATUSCODE_BADCOMMUNICATIONERROR;
    }
```
->
```c
    // We do not want to make the socket reusable.  Maybe on Windows we
    // should instead set SO_EXCLUSIVEADDRUSE, but that is not clear:
    // https://github.com/python-trio/trio/issues/928
    /*if(UA_setsockopt(newsock, SOL_SOCKET, SO_REUSEADDR,
                  (const char *)&optval, sizeof(optval)) == -1) {
        UA_LOG_WARNING(layer->logger, UA_LOGCATEGORY_NETWORK,
                       "Could not make the socket reusable");
        UA_close(newsock);
        return UA_STATUSCODE_BADCOMMUNICATIONERROR;
    }*/
```
With the change above, I get the behavior I want when starting a second server with an already used port:
```
[BOOTCAMP] [sigmund-MAC2-dev] 13:00 2.884 168661M d:\dev>temp\out\PC\opc_ua_serverD.exe 9090
[2020-06-05 13:09:56.039 (UTC+0200)] warn/server        Username/Password configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2020-06-05 13:09:56.052 (UTC+0200)] warn/userland      AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2020-06-05 13:09:56.063 (UTC+0200)] warn/network       Error binding a server socket: Only one usage of each socket address (protocol/network address/port) is normally permitted.
```

pnikiel, 2020-06-05:
a-ha. So the problem you see is Windows only then?
Could you test your idea also on Linux?

SiggyBar, 2020-06-05:
Same behavior for Ubuntu 18.04.  I do run it in WSL, let me also check on a native Linux machine.

SiggyBar, 2020-06-05:
It works as fine without my change on a native Ubuntu 18.04 machine:
```
sigmund@sverre-ubuntu1:~$ ./opc_ua_serverD 9090
[2020-06-05 13:43:18.530 (UTC+0200)] warn/server        Username/Password configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2020-06-05 13:43:18.530 (UTC+0200)] warn/userland      AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2020-06-05 13:43:18.530 (UTC+0200)] warn/network       Error binding a server socket: Address already in use
```
The same `opc_ua_serverD` binary behaves differently on Ubuntu 18.04 when running in WSL.  So this is indeed Windows specific.

paddor, 2022-03-14:
I get the same issue in an Ubuntu 20.04 VM on macOS. The `Error binding a server socket: Address already in use` gets loged but `UA_Server_run_startup()` still returns `UA_STATUSCODES_GOOD`.

xydan83, 2022-07-04:
Hi. I have the same problem on Ubuntu 21.10. I take it this issue hasn't been fixed yet?

pnikiel, 2022-07-04:
We do not see this on RedHat systems, but I can test it for you in the afternoon.

xydan83, 2022-07-04:
> We do not see this on RedHat systems, but I can test it for you in the afternoon.

It would be great if it didn't take you long. I'm really not at the release right now. 
Commit:
 e33ccf0f03f5cdf86478491fbda6a6119ffc4027 open62541 (v1.3-183-ge33ccf0f)

xydan83, 2022-07-04:
I analyzed the situation, looked at the code and it seems that this is a completely different problem (and is it a problem?). As far as I understand, total_result should accumulate results, but if at least one socket is opened, then the result is reset to zero.
This is probably correct, if at least some socket is open, then we consider that there is no error. In my case, if my port 4840 is busy on IPv4, but it can be opened on IPv6, then this is considered a successful operation, which resets the overall result.

[image]

teccheck, 2025-04-05:
We have new structures for networking due to the new event loop and connection manager, so this should be fixed now. Feel free to reopen if needed

closed 2025-04-05 by teccheck