open62541/open62541 #2510: Missing UA_EXPORT in UA_Server_getConfig(UA_Server *server) declaration

JGrothoff, 2019-02-28:
## Description
I need to get the config from the server, so i don't have to save it. Therefore an UA_EXPORT is missing for linking against open62541 library.

## Background Information / Reproduction Steps
https://github.com/open62541/open62541/blob/b41a64b31487d52e6f7d210e17a5d5e19886301c/include/ua_server.h#L1286

## Checklist

Please provide the following information:

 - [x ] open62541 Version (release number or git tag): current master
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-02-28:
Confirmed. Can you prepare a pull request?

JGrothoff, 2019-02-28:
Added the PR above.
I added another change, which is necessary for the integration in our runtime environment.

keba-estr, 2024-09-25:
@jpfr: Issue can be closed.

closed 2024-09-25 by JGrothoff