open62541/open62541 #2514: "PK - Invalid key tag or value" when reading certificate

davy7125, 2019-02-28:
Hello,

I've not been able to use a certificate for a CLIENT (this information may be important). I tried to use:
* the script here https://github.com/open62541/open62541/tree/master/tools/certs
`python.exe .\create-self_signed.py .`
* the tool opc.ua.certificategenerator.exe
`.\opc.ua.certificategenerator.exe -cmd issue -sp . -an MyApp -ks 2048`

But I get the error "[Open62541, security policy] mbedTLS returned an error: PK - Invalid key tag or value" when I use:
```
// Configuration
cc->securityMode = UA_MESSAGESECURITYMODE_SIGNANDENCRYPT;
UA_CertificateVerification_AcceptAll(&cc->certificateVerification);
cc->securityPolicyUri = UA_STRING_ALLOC("http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15");
cc->securityPoliciesSize = 1;
cc->securityPolicies = new UA_SecurityPolicy[1];
UA_StatusCode retVal = UA_SecurityPolicy_Basic128Rsa15(
  &cc->securityPolicies[0],
  &cc->certificateVerification,
  UA_BYTESTRING(m_certificate),
  UA_BYTESTRING(m_privateKey),
  &cc->logger);
```

Did I made a mistake? When reading the files I added an extra "\0".

Infinity95, 2019-03-03:
The \0 character will be ignored when setting the length of a bytestring when using `UA_BYTESTRING`.
Can you try adding 1 to the length of the produced bytestring?

davy7125, 2019-03-04:
I found the problem. I'm using C++/Cli and my mistake was to use a StreamReader instead of a basic Stream to read data. StreamReader is for text, Stream is for binary data (https://stackoverflow.com/questions/5867227/convert-streamreader-to-byte).

Here is the code I used to read the certificate that I embedded in the dll (the same is to be done with the private key). May not be optimized but it works.

```
// Get the stream (file "certificate.der" embedded in dll)
Stream^ stream = Assembly::GetExecutingAssembly()->GetManifestResourceStream("certificate.der");

// Read everything
int length = (int)stream->Length;
array<byte>^ buffer = gcnew array<byte>(length);
stream->Read(buffer, 0, length);
stream->Close();

// Convert to char* and add an extra '\0'
char * data = new char[length];
for (int i = 0; i < length; i++)
  data[i] = (char)buffer[i];
data[length] = '\0';

UA_ByteString certificate;
certificate.data = (UA_Byte*)data;
certificate.length = length + 1;
```

Then, I have been successful when using the function **UA_SecurityPolicy_Basic128Rsa15**.

closed 2019-03-04 by davy7125