open62541/open62541 #2518: Question on asynchronous Subscription/MonitoredItem_{create,modify,delete}

wbitesi, 2019-03-01:
## Description

The open62541 stack provides synchronous functions to create,modify, and delete
subscriptions and monitored items.  It does not include asynchronous pendants.

I cannot implement these asynchronous functions in my project myself, since the
synchronous operations operate on the private UA_Client structure
(ua_client_internal.h) which is not exposed.

Has the lack of asynchronous functions for subscriptions a reason due to the
internals of the client or has nobody needed/wanted this feature yet? 

## Background Information / Reproduction Steps

None

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): a82e151 
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-03-01:
Should be doable.
Marked it as a feature request.

Please comment here if you intend to work on it.

wbitesi, 2019-03-04:
I intend to work on it and make a pull-request when done.

I'll try to refactor the logic for updating the UA_Client into common functions for the async and sync functions.

wbitesi, 2019-03-06:
Pull request available: Async monitored items #2549

closed 2021-03-23 by wbitesi