open62541/open62541 #2524: Embedded Server CTT Meta-Issue

jpfr, 2019-03-02:
This Meta-Issue tracks the ongoing work for self-certification with the *Embedded UA Server* profile.

The CTT repo now contains the configuration for two certification projects:
https://github.com/open62541/open62541-ctt

- master: Has all conformance units configured and enabled that have tested green on the master branch. This is used for regression testing.
- embedded: Has all conformance units for the *Embedded Server* profile enabled. Used to track the progress.

Remaining Work:

- [x] Share the embedded configuration in the ctt repo
- [ ] Write a python-script that extracts the CTT results from the .results.xml to the Github markdown format.
- [ ] Complete the CTT configuration for the Embedded profile and the `examples/server_ctt` implementation
- [ ] Complete *Base Info GetMonitoredItems Method* (3 tests left)
- [ ] Complete *MonitorMinQueueSize_02* and *MonitorMinQueueSize_10* (2 tests left each)
- [ ] Complete *Monitor Items Deadband Filter* (2 tests left)
- [ ] Complete *Monitor Triggering* (1 test left)
- [ ] Complete *Session Base* (4 tests left)
- [ ] Complete *Subscription Minumum* (1 test left)

Please comment below if you are working on any of these items.
Or open a dedicated issue for larger subtasks.

opcua-tsn-team-kalycito, 2019-09-27:
Updating the above list with respect to current master.

Tested sha - d56732184712d67e9b13089c3518aaaf7eb0ac31(master) in the latest version of CTT (**v1.3.341.390**) we have the following items remaining:

- [ ] Base Info GetMonitoredItems Method (3 Tests left)
- [ ] Monitor MinQueueSize_02 (2 Tests left)
- [ ] Monitor Items Deadband Filter (9 Tests left)
- [ ] Monitor Triggering (36 Tests left)
- [ ] Security None (1 Test left)
- [ ] Subscription Minimum 02 (1 Test left)
- [ ] View Basic (1 Test left)

teccheck, 2025-04-05:
Certification is done now and the CCT repo is not public anymore, so I'll close this

closed 2025-04-05 by teccheck