open62541/open62541 #2533: No Client encryption APIs available in v0.3.0

bvinodkumar2008, 2019-03-04:
## Description
Seems like open62541 0.3 version (https://github.com/open62541/open62541/tree/v0.3.0) doesn't support client encryption APIs and only see server encryption APIs. I guess for the same reason, there isn't any client encryption sample made available at https://github.com/open62541/open62541/tree/v0.3.0/examples/encryption. Was wondering why this is done so?

## Checklist

Please provide the following information:

 - [ 46979d3] open62541 Version (release number or git tag):
 - [ No] Other OPC UA SDKs used (client or server): 
 - [ Ubuntu 16.04] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-03-04:
This was developed after the 0.3 feature freeze.
If you cannot upgrade to master/0.4, you'd have to find somebody that does the work for a backport.

bvinodkumar2008, 2019-03-05:
Thanks @jpfr, I will try to upgrade to latest master leading to 0.4 version.

closed 2019-03-05 by bvinodkumar2008