open62541/open62541 #2535: Compile library with visual studio 2017

klasik16, 2019-03-04:
https://stackoverflow.com/questions/54985518/build-open62541-dll-with-visual-studio-2017

jpfr, 2019-03-04:
@Pro 

This is probably related to the architecture selection.
If POSIX is selected for the architecture, then win32 will not work.
The official single-file downloads seem to be for POSIX.

There is only one solution:
Add all the architectures in the amalgamation.
And wrap them with a conditional (ifdef) so that only the detected architecture is used.

That means we need to additionally detect the architecture with macros.

jpfr, 2019-03-04:
Update:

- Changed the title to reflect the problem.
- Added to the 0.4 milestone

klasik16, 2019-03-04:
Can you write an example?

klasik16, 2019-03-05:
How actually we can use that lib in our projects for windows platform than?
Can anyone support?

StalderT, 2019-03-05:
In 0.4 the single-file seem to be for POSIX and Win32 : 

https://github.com/open62541/open62541/blob/16d2bace3fb64bb0d2f42689e07d9910aec3271d/CMakeLists.txt#L75

@klasik16 Do you use version 0.3? Can you post your VS project ?

klasik16, 2019-03-05:
i'm using 0.3 downloaded from https://github.com/open62541/open62541/releases/download/v0.3.0/open62541-win64.zip
There is Visual Studio 2017 Project:
[open62541.zip](https://github.com/open62541/open62541/files/2930258/open62541.zip)

jpfr, 2019-03-05:
Looks like a bug in Visual Studio. Let me explain.

This works. So the path with the #error macros is not active.
```c
#if defined(QUEUE_MACRO_DEBUG) || (defined(_KERNEL) && defined(DIAGNOSTIC))
#define _Q_INVALIDATE2(a) (a) = ((void *)-1)
#error this is a problem
#endif
#define _Q_INVALIDATE(a)
```

This does not work. Which is odd because the path of the first #define is not used...
```c
#if defined(QUEUE_MACRO_DEBUG) || (defined(_KERNEL) && defined(DIAGNOSTIC))
#define _Q_INVALIDATE(a) (a) = ((void *)-1)
#error this is a problem
#endif
#define _Q_INVALIDATE(a)
```

```
Severity	Code	Description	Project	File	Line	Suppression State
Warning	C4005	'_Q_INVALIDATE': macro redefinition	open62541	c:\home\projekte\open62541\tmp\open62541\open62541\open62541.c	132	
```

klasik16, 2019-03-05:
Okay. Main question is it is possible using open62541 program User interface for OPC UA server?
Second question how to do that?

I Think best tools for that world be VS 2017 and windows forms C# .net framework
Maybe someone can mail me? vsuboc.90@gmail.com
We need Working OPC UA server with UI for input data configuration....
Maybe someone world like to work together?

Pro, 2019-03-05:
By default the included architectures in the release are for win32 and posix.
If the user does not choose any specific architecture, the preprocessor automatically tries to detect if it is Windows or Posix here:
https://github.com/open62541/open62541/blob/89b41278ab8e52e48bfbf0ae1606526d447a44a8/include/ua_config.h.in#L95

So for your case you can check if that is working correctly.

@klasik16 please also try to clone the github repository, and then create a Visual Studio solution using the CMake generator.
e.g. something like this:
`cmake -DUA_ENABLE_AMALGAMATION=ON -DUA_ARCHITECTURE=win32 -DUA_BUILD_EXAMPLES=ON -G "Visual Studio 15" ..`

Then open the solution in Visual Studio and build everything. This should generate a corresponding .c/.h file.

klasik16, 2019-03-06:
Conclusion for creating .dll .lib .c and .h files is that we need use cmake to create project to able use it in VS 2017.

All right now I have .dll .lib .c and .h files.

How I suppose us them in c# windows forms project with Visual studio 2017?
We need example how create one text input and translate it throe OPC UA server.
There is some manuals on network but they are all for .c projects.

Pro, 2019-03-06:
As you already mention, now you have all you need from our side, i.e. the .dll and/or the .lib.

Basically you need the header files and the .dll, and include it in your visual studio.
If you built as I mentioned, then your examples also built correctly? So there isn't any issue with open62541 itself.

A how to on the basics of Visual Studio, especially how to include external libs is better asked on Stackoverflow, since this issue tracker is specifically a support for open62541.
E.g. see https://stackoverflow.com/q/7348675/869402

Please understand that we can not give you a tutorial on how to program or write C# apps

teccheck, 2025-04-05:
The code path for all architectures has changed substantially since the new event loop dropped, and it should work now, so I'll close this issue

closed 2025-04-05 by teccheck