open62541/open62541 #2539: Cannot build the Subscribe tutorial example?

theresa95, 2019-03-05:
I have tried to compile the PubSub tutorial examples. This includes the tutorial_pubsub_publish.c and the tutorial_pubsub_subscribe.c files. After I cloned the repository, I built the library with DUA_ENABLE_PUBSUB=ON:

    mkdir build 
    cd build
    cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_SHARED_LIBS=ON -DUA_ENABLE_AMALGAMATION=ON -DUA_ENABLE_PUBSUB=ON ..
    make

First I compiled the publish example which works:

    gcc -std=c99 -o PubSub_Publish_Example open62541.c tutorial_pubsub_publish.c

Then, I tried to compile the subscribe example in the same way like this:

    gcc -std=c99 -o PubSub_Publish_Example open62541.c tutorial_pubsub_subscribe.c
This generated lots of errors - all saying "No such file or directory(first error was for the #include "ua_architecture_base.h") Then I tried to compile it with the ua_architecture_base.h and get other missing file or directory error and so on...

I also tried to compile the whole include folder of the open62541, but I also get "No such file or directory" errors:

    gcc -std=c99 -I /home/theresa/Desktop/open62541/include -o PubSub_Subscribe_Example open62541.c tutorial_pubsub_subscribe.c

### Background Information / Reproduction Steps
I also noticed that of all examples provided by the open62541 master,
tutorial_pubsub_subscribe.c is the only example which does NOT include
the open62541.h folder. Instead it includes a few other header files:

    #include "ua_pubsub_networkmessage.h"
    #include "ua_log_stdout.h"
    #include "ua_server.h"
    #include "ua_config_default.h"
    #include "ua_pubsub.h"
    #include "ua_network_pubsub_udp.h"
    #ifdef UA_ENABLE_PUBSUB_ETH_UADP
         #include "ua_network_pubsub_ethernet.h"
    #endif
    #include "src_generated/ua_types_generated.h"
    #include <stdio.h>
    #includ <signal.h>

Of course I have read through the documentation serveral times, but it does not state how to build this example and it just describes the publish tutorial.
So how can I compile and run the subscribe tutorial example and why does the subscriber example not
include the open62541 header file?

Pro, 2019-03-05:
PubSub is currently experimental, therefore it is not completely tested. Especially it does not yet build completely if amalgamation is enabled, and therefore it is also not yet documented. Amalgamated build is only recommended for special use-cases, e.g., microcontrollers. For normal PCs it is recommended to use the non-amalgamated version.
https://github.com/open62541/open62541/blob/master/examples/CMakeLists.txt#L190

Try to disable amalgamation and then build again. Another way for you could also be to just install the library and use it through cmake:
https://open62541.org/doc/current/installing.html

Make sure that you are using the master branch.

Please re-open if the issue still exists.

closed 2019-03-05 by Pro

theresa95, 2019-03-08:
@Pro Hello,
I tried to follow the steps which are stated in the tutorial [https://open62541.org/doc/current/installing.html](url) . 

These are the steps I have taken so far:
`git clone https://github.com/open62541/open62541.git -b master`
`cd open62541`
`git submodule update --init --recursive`
`mkdir build && cd build`
`cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=FULL -UA_ENABLE_PUBSUB=ON ..`
`make`

But after I execute the make command, it gets stuck at:
`[69%] Building C object CMakeFiles/open62541-object.dir/src_generated/ua_namespace0.c.o`

I have waiting around 20 mins for it to finish but it seems to be stuck. 
The same happens withouth the build option -UA_ENABLE_PUBSUB=ON used.

What am I doing wrong here?

Pro, 2019-03-08:
Try to use `-DCMAKE_BUILD_TYPE=Debug`.

Optimizing the huge `ua_namespace0.c` file may take a lot more than 20 minutes if your machine is not that powerful.