open62541/open62541 #2540: Possible bug in UA_Client_connectInternal using a secured channel

davy7125, 2019-03-05:
I think I found a bug when reading ua_client_connect.c

1) If we read the method UA_Client_connectInternal, a check is made on "client->config.endpoint" and "client->config.userTokenPolicy" to know if we already set the endpoint.

2) Just then, a call to "UA_Client_connectTCPSecureChannel" is made and this method requires the endpoint to be set. There is for example a call to "client->config.endpoint.serverCertificate" which would result in a null pointer exception.

3) Back to the function "UA_Client_connectInternal", the endpoint is selected with "selectEndpoint" if it's not been set (step 1 gives the result).

It's like a snake biting its tail. We need to know the endpoint to get a secured channel but the secured channel is needed to get the endpoint?

jpfr, 2019-03-05:
The first SecureChannel (for GetEndpoints) is not encrypted.
All servers have endpoints for unencryptedSecureChannels.
Specifically for GetEndpoints.

davy7125, 2019-03-05:
What I suggested is not working but there is definitely a bug. UA_Client_connectTCPSecureChannel cannot rely on "client->config.endpoint" since possibly we don't know it at this stage.

It is possible to already remove 2 occurrences of the endpoint:
- client->config.endpoint.securityMode becomes client->config.securityMode
- client->config.endpoint.securityPolicyUri becomes client->config.securityPolicyUri

But there is still "client->config.endpoint.serverCertificate". Is it possible to read it in another way?

davy7125, 2019-03-06:
I finally found that the existing code is working. It was linked to #2538: when analyzing the function "getSecurityPolicy" I found out that I needed to add the "None" security policy otherwise the library was not able to list the endpoints. Added logs helped me to debug this, that's why I created the pull request #2545.

closed 2019-03-06 by davy7125